RobotLocomotion/director #21: can't see the lidar points (from lcmgl) in the drake viewer on my mac

RussTedrake, 2015-07-15:
```
cd drakes/examples/Quadrotor;
Quadrotor.runOpenLoop(); 
```

the lidar returns show up in my libbot viewer but not director.  maybe they're just too small?

requires this fix to work:
https://github.com/RobotLocomotion/drake/pull/1181

patmarion, 2015-07-15:
attached screenshot from linux.  can confirm the lcmgl is working on that platform.

do you have lcmgl working in any examples on your mac?  There is an issue with mac opengl x11 linking into libbot that prevents it from working without a libbot patch, I'll try to find the git issue reference for that one.

if you press f1 it opens the scene browser.  see in my screenshot, you can see the lcmgl object named 'lidar' shows up.  If you have that showing up, but it's not displayed, then it is likely the opengl x11 bug (which is resolvable, especially if we remove lcmgl from libbot so it no longer links to x11)

[image]

patmarion, 2015-07-15:
This is the libbot patch:

diff --git a/bot2-lcmgl/src/bot_lcmgl_
render/CMakeLists.txt b/bot2-lcmgl/src/bot_lcmgl_render/CMakeLists.txt
index f4a7579..d70da90 100644
--- a/bot2-lcmgl/src/bot_lcmgl_render/CMakeLists.txt
+++ b/bot2-lcmgl/src/bot_lcmgl_render/CMakeLists.txt
@@ -5,7 +5,11 @@ file(GLOB h_files *.h)
 
 add_library(bot2-lcmgl-renderer SHARED ${c_files})
 
-set(REQUIRED_LIBS bot2-vis lcmtypes_bot2-lcmgl gl)
+find_package(OpenGL REQUIRED)
+find_package(GLUT REQUIRED)
+target_link_libraries(bot2-lcmgl-renderer ${OPENGL_LIBRARIES} ${GLUT_LIBRARIES})
+
+set(REQUIRED_LIBS bot2-vis lcmtypes_bot2-lcmgl)
 pods_use_pkg_config_packages(bot2-lcmgl-renderer ${REQUIRED_LIBS})

RussTedrake, 2015-07-15:
would you be so kind as to pull request that against the libbot pod?

patmarion, 2015-07-15:
could you try the new libbot pull request, just pull and make.  It fixes the opengl link for mac, and this may resolve your problem.

RussTedrake, 2015-07-15:
very nice.  had to also make clean in directory, but that worked.  you're awesome.

closed 2015-07-15 by RussTedrake