RobotLocomotion/director #24: compiling on ubuntu 12.04

RussTedrake, 2015-07-17:
still have this issue (perhaps due to an older version of numpy?)

[100%] Building C object src/python/extensions/transformations/CMakeFiles/_transformations.dir/transformations.c.o
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleArray':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:780:16: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:780:16: note: each undeclared identifier is reported only once for each function it appears in
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_AnyDoubleArray':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:796:20: error: 'NPY_ARRAY_ALIGNED' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleArrayOrNone':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:813:20: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleMatrix44':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:828:16: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleMatrix44Copy':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:850:16: error: 'NPY_ARRAY_ENSURECOPY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:850:16: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleVector3':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:873:16: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleVector4':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:895:16: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleVector4Copy':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:917:16: error: 'NPY_ARRAY_ENSURECOPY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:917:16: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'PyConverter_DoubleVector3OrNone':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:943:20: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c: In function 'py_inverse_matrix':
/tmp/drake-master/director/src/python/extensions/transformations/transformations.c:3154:31: error: 'NPY_ARRAY_IN_ARRAY' undeclared (first use in this function)
make[6]: *** [src/python/extensions/transformations/CMakeFiles/_transformations.dir/transformations.c.o] Error 1
make[5]: *** [src/python/extensions/transformations/CMakeFiles/_transformations.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [src/ddapp-stamp/ddapp-build] Error 2

patmarion, 2015-07-17:
The compile error is coming from transformations.c.  This is a 3rd party extension module that I added last week.  Thanks for catching the compile error!

transformations.c requires numpy min version of 1.7, but Ubuntu 12.04 provides only 1.6.
Since this transformations.c extension module is optional, I think the best fix for now is to detect which numpy version is available and disable transformations.c if the version number cannot be satisfied.

If you'd like to work around the issue for now, edit src/python/extensions/CMakeLists.txt and comment out:

    add_subdirectory(transformations)

Or, install a newer version of numpy.  I think you may be able to do that with:

    $ sudo apt-get install python-pip
    $ sudo pip install numpy

One way to get the available numpy version is with this command (assuming numpy is installed):

    $ python -c "import numpy; print(numpy.__version__)"

When I added transformations.c, I added some quick cmake code to locate the numpy include directory.  I should do a more comprehensive FindNumpy that checks the version number, and possible use a better method to locate the header directory, such as:

    $ python -c "import numpy; print(numpy.get_include())"

RussTedrake, 2015-07-17:
I'll wait for the correct fix to get pushed.  I'm not blocked (it's only a build serve that failed to compile the director).  Thanks

patmarion, 2015-07-17:
should be fixed now.  It will check the numpy version and if it's too old then you'll get a cmake warning but no compiler error.

patmarion, 2015-07-17:
for drake-distro, the new submodule sha1 for director should be f44045f should I make a pull request?

RussTedrake, 2015-07-20:
done.

patmarion, 2015-07-28:
I think this is resolved now.

closed 2015-07-28 by patmarion

AndyMarchese, 2015-10-01:
An interesting thing happened to me surrounding this issue:
On Ubuntu 12.04, I had unknowingly upgraded numpy to version 1.9. The version check Pat provided above confirmed this. This setup generated the same make error Russ reported in the original post. After uninstalling numpy 1.9 and defaulting back to 1.6, the issue was gone. I'm not sure how frequent this situation would arise, but maybe checking the numpy version number against the Ubuntu version would cover all bases?

patmarion, 2015-10-02:
Hmm, it could have used the include dir for numpy 1.6 even though it detected numpy 1.9 was installed.  But I'm not sure how that would happen, it's suppose to include the exact version that is detected.  If you still had numpy 1.9 installed, I'd ask you check your CMakeCache.txt file so I could see what might be going wrong:

$ grep NUMPY externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build/CMakeCache.txt