RobotLocomotion/director #30: Segfault upon loading ply file

iamwolf, 2015-07-23:
Backtrace:
```
ERROR: In /build/buildd/vtk-5.8.0/IO/vtkPLYReader.cxx, line 131
vtkPLYReader (0x64b2320): Cannot read geometry

Program received signal SIGSEGV, Segmentation fault.
0x00007fffef68459b in vtkPLY::equal_strings(char const*, char const*) () from /usr/lib/libvtkIO.so.5.8
(gdb) backtrace
#0  0x00007fffef68459b in vtkPLY::equal_strings(char const*, char const*) () from /usr/lib/libvtkIO.so.5.8
#1  0x00007fffef684607 in vtkPLY::find_element(PlyFile*, char const*) () from /usr/lib/libvtkIO.so.5.8
#2  0x00007fffef687cd7 in vtkPLYReader::RequestData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) () from /usr/lib/libvtkIO.so.5.8
#3  0x00007fffeef056e0 in vtkExecutive::CallAlgorithm(vtkInformation*, int, vtkInformationVector**, vtkInformationVector*) () from /usr/lib/libvtkFiltering.so.5.8
#4  0x00007fffeeef75a7 in vtkDemandDrivenPipeline::ExecuteData(vtkInformation*, vtkInformationVector**, vtkInformationVector*) () from /usr/lib/libvtkFiltering.so.5.8
#5  0x00007fffeeefa461 in vtkDemandDrivenPipeline::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) () from /usr/lib/libvtkFiltering.so.5.8
#6  0x00007fffef04c921 in vtkStreamingDemandDrivenPipeline::ProcessRequest(vtkInformation*, vtkInformationVector**, vtkInformationVector*) () from /usr/lib/libvtkFiltering.so.5.8
#7  0x00007fffeeef8e4d in vtkDemandDrivenPipeline::UpdateData(int) () from /usr/lib/libvtkFiltering.so.5.8
#8  0x00007fffef04931e in vtkStreamingDemandDrivenPipeline::Update(int) () from /usr/lib/libvtkFiltering.so.5.8
#9  0x00007fffb358410d in ?? () from /usr/lib/libvtkFilteringPythonD.so.5.8
#10 0x00007fffec6ce0d4 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#11 0x00007fffec6cf54d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#12 0x00007fffec6cddd8 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#13 0x00007fffec6ce059 in PyEval_EvalFrameEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#14 0x00007fffec6cf54d in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#15 0x00007fffec7046d0 in ?? () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#16 0x00007fffec670d43 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#17 0x00007fffec6e9577 in PyEval_CallObjectWithKeywords () from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
#18 0x00007ffff50240bb in PythonQtSignalTarget::call(_object*, PythonQtMethodInfo const*, void**, bool) () from /home/wxm/MSc_Project/drc/software/build/lib/libPythonQt.so
#19 0x00007ffff5024156 in PythonQtSignalTarget::call(void**) const () from /home/wxm/MSc_Project/drc/software/build/lib/libPythonQt.so
#20 0x00007ffff5024b9b in PythonQtSignalReceiver::qt_metacall(QMetaObject::Call, int, void**) () from /home/wxm/MSc_Project/drc/software/build/lib/libPythonQt.so
#21 0x00007ffff5c9fa78 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#22 0x00007ffff5c9f87a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#23 0x00007ffff6d93a62 in QAction::triggered(bool) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#24 0x00007ffff6d95433 in QAction::activate(QAction::ActionEvent) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#25 0x00007ffff6d9558c in QAction::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#26 0x00007ffff6d99e2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#27 0x00007ffff6da04a0 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#28 0x00007ffff5c8b4dd in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#29 0x00007ffff6dcb2a6 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#30 0x00007ffff6dcb3dc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#31 0x00007ffff6da1773 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#32 0x00007ffff5c8b4dd in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#33 0x00007ffff6e39027 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#34 0x00007ffff6e393c9 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#35 0x00007ffff6e13417 in QApplication::x11ProcessEvent(_XEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#36 0x00007ffff6e3bb32 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#37 0x00007fffee398e04 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#38 0x00007fffee399048 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#39 0x00007fffee3990ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#40 0x00007ffff5cb87a1 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#41 0x00007ffff6e3bbe6 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#42 0x00007ffff5c8a0af in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#43 0x00007ffff5c8a3a5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#44 0x00007ffff5c8fb79 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#45 0x0000000000400be8 in main ()
```

The ply in question was written by PCL and can be opened without problems in MeshLab. I've uploaded it [here](https://gist.github.com/iamwolf/be2ca51af46849e9744b). The problem seems to come from vtk directly, but perhaps we can handle this more gracefully. Should I perhaps use a different/newer vtk version?

patmarion, 2015-07-28:
Unfortunately, the segfault is in the vtkPLYReader c++ code and cannot be avoided.  I was able to repeat the crash on my Ubuntu 14.04 (vtk 5.8) but I did not test Mac (vtk 5.10).

I was able to get the file to load successfully by making a small modification to the ply file.  The ply file does not declare any information about faces (only vertices) so I just added a declaration for 0 faces.  In the ply header, after the line "property uchar blue" I added these two lines:

    element face 0
    property list uchar int vertex_indices

Then VTK loads the file without problem.  However, the file does not display as you would expect.  The data contains vertices but no elements or cells.  For VTK to draw the data as a point cloud it needs to have point cells.  You can add point cells to a data object from the python console:

pd = om.findObjectByName('octomap_test.ply').polyData
vtk.vtkPCLConversions.AddVertexCells(pd)

[image]

I have not used the assimp library, but it's possible that bringing in a new dependency like assimp would give us access to more data file formats and we could replace the vtkPLYReader.  For more information about PLY file format, see:

http://paulbourke.net/dataformats/ply/

iamwolf, 2015-07-31:
Thanks for investigating and the explanation. Since it's coming from VTK itself, should we declare this as a wontfix and close the issue?

patmarion, 2015-07-31:
Yes, unfortunately I think this is a wontfix.  But if more of these issues come up then maybe we should add the asset library for i/o of these formats.

closed 2015-07-31 by patmarion

rennaisance-dangelo, 2016-02-23:
Hi Pate,
I am kind of late in this discussion, but I face a similar problem. I used vtkPLYReader in my code with a .ply file which I converted using Point Clouds Library (PCL) from .pcd to .ply. Now the issue is that due to the process of conversion I lost the faces' values in my .ply file. Due to this, upon using the vtkPLYReader I receive an error of segmentation fault as mentioned above. I saw your comment above in which you mentioned a python script to generate the faces in the .ply file. The script you mentioned is as under:

`pd = om.findObjectByName('octomap_test.ply').polyData
vtk.vtkPCLConversions.AddVertexCells(pd)`

Out of curiosity and being a newbie I tried to clone director's repo and tried to compile it. However, due to very less accompanying documentation, I am not able to find out the "om" module that you have mentioned in the above script. Would it be possible that you could let me know how to find this function among the python scripts in director and use it? I will be really grateful.

patmarion, 2016-02-23:
The code I posted only works in the openhumanoid's version of director.  Is that the version you compiled?  The above code should run in the Python console of director.  om is an abbreviation for objectmodel, director automatically imports the objectmodel.py module and aliases it with om.

If you downloaded director stand alone repo and compiled it, then you don't have pcl or the main director program, you just have the directorPython program.  If that's what you are using then the process will be different.

rennaisance-dangelo, 2016-02-24:
Thank you for your reply. I am afraid that the code that I have compiled is from the stand alone repo (from Robot Locomotion). I checked the fork of 'director' on the openhumanoids repo, but it seems like the code is same there too. Do you mean to say that this script works only with 'director' version which is available with 'oh-distro' on the openhumanoids repo?

I have pcl libraries pre-installed on my computer, is there a way to use them along with the version of 'director' that has been compiled as stand alone? Which means that I have directorPython program.