RobotLocomotion/director #37: Move RobotPoseGUI and create unit test

iamwolf, 2015-07-31:
Follow up on https://github.com/mitdrc/drc/pull/2077

 1. Move RobotPoseGUI into the director repository
 1. Create unit test for RobotPoseGUIWrapper

Re (1): @patmarion where to put this tool best? Is creating a folder underneath ``src/python`` for ``tools`` a good idea? Any other place to put it?

patmarion, 2015-07-31:
We used to use the RobotPoseGUI as a standalone program, launched from the command line.  I think people mostly use it now by launching it from within the director.  I wouldn't mind if we dropped support for it as a standalone program, which would simplify its maintenance and then it could be renamed something like  src/python/ddapp/robotposturepanel.py and it could be launched standalone using ddConsoleApp -m ddapp.robotposturepanel and if we wanted to install a shell script to launch it with a simpler command that's also possible.

iamwolf, 2015-07-31:
Great, thanks, I'd do this and make a PR in a bit. It's also great because as of now it would be hard to test as the ddConsoleApp main window isnt of type ddMainWindow and thus fails.

mauricefallon, 2015-07-31:
I don't know @patmarion , maybe its time for stickyhands and the n-curses tool to make a comeback. ;)

Seriously... +1  its not used outside of the designer.

iamwolf, 2015-07-31:
I added checking whether the PoseGUI can be opened to the teleopPanel unit test (happy to create a stand alone unit test as well - thought it'd fit in there given that it's queried from the teleopPanel).

Also happy work on moving it into the director, perhaps at a later stage though since it will need a few changes along with it.

mauricefallon, 2016-01-21:
Can this be closed , after https://github.com/RobotLocomotion/director/pull/164 ?

patmarion, 2016-01-21:
Yes meant to close this, thanks!

closed 2016-01-21 by patmarion