RobotLocomotion/director #39: missing FindVTK.cmake on drake001 (mac)

RussTedrake, 2015-08-01:
```
[ 78%] Performing configure step for 'ddapp'
loading initial cache file /tmp/drake-distro/director/distro/pods/drake-distro/pod-build/tmp/ddapp-cache.cmake
CMake Error at src/vtk/CMakeLists.txt:1 (find_package):
  By not providing "FindVTK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VTK", but
  CMake did not find one.

  Could not find a package configuration file provided by "VTK" with any of
  the following names:

    VTKConfig.cmake
    vtk-config.cmake

  Add the installation prefix of "VTK" to CMAKE_PREFIX_PATH or set "VTK_DIR"
  to a directory containing one of the above files.  If "VTK" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
```
```
drake001% cmake --version
cmake version 3.2.3
```

RussTedrake, 2015-08-01:
looks like vtk might not have been installed properly.  (remember I started it at the conference, then shut my laptop? :)  will reopen if it's still a problem after this.

closed 2015-08-01 by RussTedrake

naveenoid, 2016-01-20:
Im having identical issues on a linux build

[ 78%] Performing configure step for 'ddapp'
loading initial cache file /home/naveenoid/Workspace/Drake/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/ddapp-cache.cmake
CMake Error at /usr/share/cmake-2.8/Modules/FindVTK.cmake:135 (message):
  VTK not found.  Set the VTK_DIR cmake cache entry to the directory
  containing VTKConfig.cmake.  This is either the root of the build tree, or
  PREFIX/lib/vtk for an installation.  For VTK 4.0, this is the location of
  UseVTK.cmake.  This is either the root of the build tree or
  PREFIX/include/vtk for an installation.
Call Stack (most recent call first):
  src/vtk/CMakeLists.txt:1 (find_package)

Am I missing something fundamental in my make options? I used the default set and then I switched to using  WITH_ALL_PUBLIC_EXTERNALS as ON

patmarion, 2016-01-20:
Can you confirm you have vtk installed?  Try these:

sudo apt-get install libqt4-dev libvtk5-dev libvtk5-qt4-dev python-vtk python-numpy

If those are installed and you still get the error message, can you confirm this file exists:

/usr/lib/vtk-5.8/VTKConfig.cmake

And what OS version do you have?

naveenoid, 2016-01-20:
I can confirm that file exists (VTKConfig.cmake). My OS is a Linux Mint 17 (kernel : 3.13.0-24-generic x86_64 (64 bit)). I had manually installed vtk6 after the first failure and your instruction corrected that. However I have another build error now : 

  5%] Performing build step for 'cdd_project'
Making all in lib-src
Making all in src
Making all in lib-src-gmp
libtool: compile:  gcc -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE=\"cddlib\" -DVERSION=\"0.94\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DSTDC_HEADERS=1 -I. -I/usr/local/include -DGMPRATIONAL -g -O2 -MT cddcore.lo -MD -MP -MF .deps/cddcore.Tpo -c cddcore.c  -fPIC -DPIC -o .libs/cddcore.o
In file included from cdd.h:18:0,
                 from cddcore.c:14:
cddmp.h:31:18: fatal error: gmp.h: No such file or directory
  #include "gmp.h"
                  ^

FYI from related issue discussions, I think this info is also relevant, my Cmake version is 2.8.12.2