RobotLocomotion/director #41: cairo lib issue suddenly appeared on mac

RussTedrake, 2015-08-05:
```
Traceback (most recent call last):
  File "/usr/local/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/local/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/Users/russt/drake-distro/build/lib/python2.7/dist-packages/ddapp/drakevisualizer.py", line 5, in <module>
    import ddapp.applogic as app
  File "/Users/russt/drake-distro/build/lib/python2.7/dist-packages/ddapp/applogic.py", line 5, in <module>
    import ddapp.vtkAll as vtk
  File "/Users/russt/drake-distro/build/lib/python2.7/dist-packages/ddapp/vtkAll.py", line 2, in <module>
    from vtkDRCFiltersPython import *
ImportError: dlopen(/Users/russt/drake-distro/build/lib/python2.7/dist-packages/ddapp/vtkDRCFiltersPython.so, 2): Symbol not found: _cairo_xlib_surface_create
  Referenced from: /usr/local/lib/libgdk-x11-2.0.0.dylib
  Expected in: /usr/local/lib/libcairo.2.dylib
 in /usr/local/lib/libgdk-x11-2.0.0.dylib
```

patmarion, 2015-08-05:
I would use homebrew to uninstall vtk and then reinstall vtk.  I suspect that the gtk libs have been recompiled and now the vtk libs are outdated.

brew uninstall vtk5
brew install vtk5 --with-qt

RussTedrake, 2015-08-05:
will try.  
but it turns out i got a similar error running the libbot viewer.  that wouldn’t be explained by the vtk issued, would it?

> On Aug 5, 2015, at 5:46 PM, Pat Marion <notifications@github.com> wrote:
> 
> brew uninstall vtk5
> brew install vtk5 --with-qt
> 
>

patmarion, 2015-08-05:
Yeah, that wouldn't have anything to do with vtk.  Maybe you need to reinstall cairo.  My interpretation of the error message is that libgdk-x11-2.0.0.dylib is expecting the symbol _cairo_xlib_surface_create to be provided by libcairo.2.dylib.  So perhaps at the time gdk was compiled the libcairo had the symbol, but since then libcairo has been compiled without X11 support?

patmarion, 2015-08-05:
I'm not up to date with the gtk/x11 issues on new homebrew, this is related I suppose?

Do you have plans to eventually drop support for the libbot viewer in order to lose the gtk with X11 dependency (and only provide the director + lcmgl)?

RussTedrake, 2015-08-05:
yes - happy to stop with libbot.  the deprecated warnings are scary and it’s not nearly as nice.  the only advantage is that it is smaller and easier to compile.

patmarion, 2015-08-08:
@RussTedrake resolved?

rdeits, 2015-08-08:
By the way, this was the exact same error I was getting, which was resolved by installing the older versions of gtk+/cairo/pango https://github.com/RobotLocomotion/libbot/issues/3#issuecomment-128793359

RussTedrake, 2015-08-09:
> By the way, this was the exact same error I was getting, which was resolved by installing the older versions of gtk+/cairo/pango RobotLocomotion/libbot#3 (comment) <https://github.com/RobotLocomotion/libbot/issues/3#issuecomment-128793359>

not resolved yet, but agree that it’s the same issue as the libbot one.

patmarion, 2015-08-10:
After pull request RobotLocomotion/drake#1245 is merged it will be possible to simply turn off bot2-vis in the libbot build and the director will no longer have any references to libraries that are causing issues.

You can use this command from the drake top-level dir to disable bot2-vis in lcmgl:

cmake -DUSE_BOT_VIS=OFF externals/libbot/bot2-lcmgl/pod-build
make

RussTedrake, 2015-08-18:
this is resolved.

closed 2015-08-18 by RussTedrake