RobotLocomotion/director #43: Citation information

iamwolf, 2015-08-10:
@patmarion Do you have a pre-prepared bibtex snippet on how to best cite the director? And would you like to add this citation information to the readme?

Thanks,
Wolf

patmarion, 2015-08-10:
No, I don't have bibtex for citing the director.  The closest thing that has been published might be the JFR paper on affordance based planning.

https://scholar.google.com/citations?view_op=view_citation&hl=en&user=BqV8LaoAAAAJ&citation_for_view=BqV8LaoAAAAJ:Zph67rFs4hoC

If you'd like to cite the source code, perhaps a cite to the github url?

cdarpino, 2015-08-10:
I encourage you to have one recommended citation form for the Director (I'll use it). For example, Drake has the following:

>>    Citing Drake:
    If you would like to cite Drake in your academic publications, we suggest the following BibTeX citation:
    @misc{drake,
    author = "Russ Tedrake",
    title = "Drake: A planning, control, and analysis toolbox for nonlinear dynamical systems",
    year = 2014,
    url = "http://drake.mit.edu"
   }

I think you can do something similar using the github link.

patmarion, 2015-08-10:
Following the drake style citation, it could be:

<pre>

@misc{director,
author = "Pat Marion",
title = "Director: A robotics interface and visualization framework",
year = 2015,
url = "http://github.com/RobotLocomotion/director"
}

</pre>

I don't like that the citation gives me sole credit for the software since it has seen many contributors, but it's fair to say I am the primary author and maybe in the future there will be a paper to cite that could be used to update this.  What do you guys think?

cdarpino, 2015-08-10:
In my view this works well! I'll be happy to use it.
It can be augmented when the paper is out (keeping the url).
Also, the git history shows the contributions.

RussTedrake, 2015-08-10:
I feel the same about drake.  It made more sense when i posted it, but now feels silly.

commit d0e09c55 (2015-08-17): Merge pull request #53 from ipab-slmc/citation

Citation info, closes #43
closed 2015-08-17 by patmarion