RobotLocomotion/director #56: compile error on parallel build

RussTedrake, 2015-08-18:
```
[ 85%] Building CXX object src/app/CMakeFiles/ddApp.dir/ddViewMenu.cpp.o
[ 85%] Building CXX object src/app/CMakeFiles/ddApp.dir/vtkSimpleActorInteractor.cxx.o
[ 86%] Building CXX object src/app/CMakeFiles/ddApp.dir/vtkTDxInteractorStyleCallback.cxx.o
[ 86%] Building CXX object src/app/CMakeFiles/ddApp.dir/moc_ddLCMSubscriber.cxx.o
[ 87%] Building CXX object src/app/CMakeFiles/ddApp.dir/moc_ddLCMThread.cxx.o
[ 87%] Building CXX object src/app/CMakeFiles/ddApp.dir/ddLCMThread.cpp.o
Linking CXX shared library ../../lib/libddApp.dylib
ld: library not found for -lQVTK
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[10]: *** [lib/libddApp.dylib] Error 1
make[9]: *** [src/app/CMakeFiles/ddApp.dir/all] Error 2
make[8]: *** [all] Error 2
make[7]: *** [src/ddapp-stamp/ddapp-build] Error 2
make[6]: *** [CMakeFiles/ddapp.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
make[2]: *** [CMakeFiles/director.dir/all] Error 2
```

this is on the drake005 running with -j 12.  is it possible that there is a dependency missing?  the error seems random and the serial build (has) worked fine on the same machine.

patmarion, 2015-08-18:
Looks like drake005 has homebrew installed vtk, not vtk5.

try:

brew uninstall vtk
brew install vtk5 --with-qt

patmarion, 2015-08-18:
I can't imagine it would be a random failure though, I think it would always fail due to this.

patmarion, 2015-08-18:
And while you're at it, signal-scope dependency for qwt is missing.  Sorry, signal-scope may not have an install_prereqs.sh file yet.  I'll add that.

On drake005 please run:

brew install qwt

RussTedrake, 2015-08-18:
now i'm running the vtk5 install that you've put above.  but i swear it compiled after the previous fix...
(sorry for being flakey; trying to maintain too many machines)

RussTedrake, 2015-08-18:
i think that's resolved it.  so the remaining request is to update the install_prereqs so I don't hit this next time.  :)

closed 2015-08-18 by RussTedrake

patmarion, 2015-08-18:
install_prereqs.sh correctly lists vtk5.  I think I said vtk and not vtk5 in a github comment thread, and that was a typo.

patmarion, 2015-08-18:
btw, I think @rdeits was working on updating the install_prereqs logic, but not sure if that has been added yet.  The director has its install_prereqs.sh script located in a subdirectory, so it might not be executed correctly at the moment.  It is located at externals/director/distro/pods/drake-distro/install_prereqs.sh

rdeits, 2015-08-18:
That update to the logic is coming in #1254 The effect should be that install_prereqs will recurse into whatever the SOURCE_DIR property is set to for each external project.

edwardtung, 2015-10-18:
hi
i encountered the same problem and try the same

$ brew uninstall vtk
Error: No such keg: /usr/local/Cellar/vtk
$ brew install vtk5 --with-qt
==> Installing vtk5 from homebrew/homebrew-science
Error: You must `brew link pyqt' before homebrew/science/vtk5 can be installed
$ brew link pyqt
Linking /usr/local/Cellar/pyqt/4.11.4... Error: Permission denied - /usr/local/lib/python2.7/site-packages/PyQt4

what should i do?

atefemoosavi, 2015-11-22:
I have the same problem when compiling drake-distro on ubuntu, and no same vtk is exist !!!
```
/bin/sh: 1: ./depth_compose.cxx: not found
make[10]: *** [src/vtk/DRCFilters/depth_compose.cxx] Error 127
make[9]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
make[8]: *** [all] Error 2
make[7]: *** [src/ddapp-stamp/ddapp-build] Error 2
make[6]: *** [CMakeFiles/ddapp.dir/all] Error 2
make[5]: *** [all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

any suggestion?

patmarion, 2015-11-23:
It looks like you do not have the correct version of VTK installed.  Can you confirm that you've followed the drake instructions to install prereqs?  Can you confirm you have vtk-5.8 installed?  This is the apt-get install command for compiling director on Ubuntu 14.04:

sudo apt-get install cmake libvtk5-qt4-dev python-dev python-vtk python-numpy

Afterward, this file should exist:

  /usr/lib/vtk-5.8/VTKConfig.cmake

atefemoosavi, 2015-11-23:
Yes, I followed the instruction step by step and I have all of above packages, also have: /usr/lib/vtk-5.8/VTKConfig.cmake
but the problem still exists !
but as a point, I don't know which section in make options should disable?

patmarion, 2015-11-23:
Maybe you have multiple versions of VTK installed and it is building with the wrong one?  From the top-level drake directory, try running this command post the output:

$ grep VTK externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build/CMakeCache.txt 

You should see:

VTK_DIR:PATH=/usr/lib/vtk-5.8

If it has a different value, try editing the CMakeCache.txt file ( externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build/CMakeCache.txt  ) and modify the line with VTK_DIR to be as above.  Then try running make again.

patmarion, 2015-11-23:
To disable director, in make options, disable the option WITH_DIRECTOR.  With director disabled you will have more limited visualization capabilities.

atefemoosavi, 2015-11-23:
Thank you. it resolved, but now I have another compiling problem, if I enable the option WITH_EIGEN I have this error:
```
[ 80%] Performing build step for 'eigen'
[ 12%] Performing download step (download, verify and extract) for 'eigen'
-- downloading...
     src='https://bitbucket.org/eigen/eigen/get/3.2.6.tar.bz2'
     dst='/home/atefe/new-drake-distro/externals/eigen/pod-build/eigen-prefix/src/3.2.6.tar.bz2'
     timeout='none'
CMake Error at eigen-stamp/download-eigen.cmake:21 (message):
  error: downloading 'https://bitbucket.org/eigen/eigen/get/3.2.6.tar.bz2'
  failed

    status_code: 1
    status_string: "Unsupported protocol"
    log: Protocol "https" not supported or disabled in libcurl

  Closing connection -1
```
and if I disable it, having this error:
```
 Could not find a package configuration file provided by "Eigen3" (requested
  version 3.2) with any of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.
```

patmarion, 2015-11-23:
Have you installed a non-standard version of cmake?  What do you get if you run:

$ which cmake
$ cmake --version

It appears that the version of cmake you are using was built without SSL support, so it is unable to download eigen from the "https://bitbucket.org" url.  You could attempt to manually download https://bitbucket.org/eigen/eigen/get/3.2.6.tar.bz2 and place it at the destination location /home/atefe/new-drake-distro/externals/eigen/pod-build/eigen-prefix/src/3.2.6.tar.bz2  but I'm not sure if that will actually work.

huzijian1996, 2021-06-21:
> Have you installed a non-standard version of cmake? What do you get if you run:
>
> $ which cmake
> $ cmake --version
>
> It appears that the version of cmake you are using was built without SSL support, so [...]

I tried this but failed. I can not access the link https://bitbucket.org/eigen/eigen/get/3.3.3.tar.gz . I downloaded eigen manually and tried to comment out some code to avoid a second download, but failed. And my cmake version is 3.5.1.Hope u can solve my problem.Thanks a lot.