RobotLocomotion/director #57: yet another compile error

RussTedrake, 2015-08-18:
on drake005

```
-- Build files have been written to: /tmp/drake-1191-8727/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build
[ 93%] Performing build step for 'ddapp'
[  1%] Generating depth_compose.cxx
/bin/sh: ./depth_compose.cxx: No such file or directory
make[6]: *** [src/vtk/DRCFilters/depth_compose.cxx] Error 127
make[5]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [src/ddapp-stamp/ddapp-build] Error 2
make[2]: *** [CMakeFiles/ddapp.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

patmarion, 2015-08-18:
Thanks.  I'm logged in and looking into it.

patmarion, 2015-08-18:
I think this is also due to the same homebrew dependency issue referenced in https://github.com/RobotLocomotion/director/issues/56#issuecomment-132323338

RussTedrake, 2015-08-18:
that resolved it

closed 2015-08-18 by RussTedrake