RobotLocomotion/director #60: Director really really really wants to use Qt version 4.8.6 (but I have 4.8.7)

rdeits, 2015-08-19:
As of today, the director build is failing with the following error:

```
[  6%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/depth_compose.cxx.o
gnumake[10]: *** No rule to make target `/usr/local/Cellar/qt/4.8.6/lib/QtGui.framework', needed by `lib/libvtkDRCFilters.dylib'.  Stop.
gnumake[9]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
gnumake[8]: *** [all] Error 2
gnumake[7]: *** [src/ddapp-stamp/ddapp-build] Error 2
gnumake[6]: *** [CMakeFiles/ddapp.dir/all] Error 2
gnumake[5]: *** [all] Error 2
gnumake[4]: *** [all] Error 2
gnumake[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
gnumake[2]: *** [CMakeFiles/director.dir/all] Error 2
gnumake[1]: *** [all] Error 2
make: *** [all] Error 2
```

which makes sense, because I don't have Qt 4.8.6; I have 4.8.7. The configuration process seems to know that:

```
-- Looking for QT_MAC_USE_COCOA
-- Looking for QT_MAC_USE_COCOA - found
-- Found Qt4: /usr/local/bin/qmake (found version "4.8.7")
```

but no matter what I do, the vtk filters still try to link against a non-existent version 4.8.6. My `qmake --version` returns:

```
± qmake --version
QMake version 2.01a
Using Qt version 4.8.7 in /usr/local/Cellar/qt/4.8.7/lib
```

which is correct. 

I've done a clean build in director, manually removed director/distro/pods/drake-distro/pod-build, and even nuked the entire drake/pod-build folder, to no avail. 

Any ideas?

patmarion, 2015-08-19:
Is it possible that you homebrew installed vtk while qt v4.8.6 was on your system?  I suspect this might fix it:

brew uninstall vtk5
brew install vtk5 --with-qt

The problem is that when the director uses vtk, it imports vtk's libraries and those reference non-existent qt v4.8.6.

rdeits, 2015-08-19:
That's probably the case. I'll reinstall vtk5 and see what happens.

patmarion, 2015-08-24:
@rdeits resolved?

rdeits, 2015-08-24:
Possibly, but I'm now getting QT errors when building signal-scope. But I'm not entirely sure that the vtk5 install completed successfully, so I'm going to retry that.

patmarion, 2015-08-24:
For signal-scope, you need to brew install qwt.

rdeits, 2015-08-24:
It works! Thanks

closed 2015-08-24 by rdeits