RobotLocomotion/director #71: 'Segment Table' not working

cdarpino, 2015-09-14:
I think the 'segment table' functionality is currently broken. I get the error below. Any thoughts? 

[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.Generic Warning: In /build/buildd/vtk-5.8.0/Common/vtkMath.cxx, line 553
Unable to factor linear system

Generic Warning: In /build/buildd/vtk-5.8.0/Common/vtkMath.cxx, line 553
Unable to factor linear system

Generic Warning: In /build/buildd/vtk-5.8.0/Common/vtkMath.cxx, line 553
Unable to factor linear system

Generic Warning: In /build/buildd/vtk-5.8.0/Common/vtkMath.cxx, line 553
Unable to factor linear system

...

patmarion, 2015-09-14:
I think the first error message from pcl::VoxelGrid is relevant.  Perhaps your input point cloud has very large extents, so the VoxelGrid filter fails and passes through an empty point cloud.  The table segmentation algorithm could be updated to detect an empty point cloud, but as it is it just continues and spits out more error messages.

patmarion, 2015-09-14:
A reasonable fix might be to crop the input point cloud by several meters centered around the input search position.  For example:

<pre>
+    polyData = cropToSphere(polyData, searchPoint, 5)
    polyData = applyVoxelGrid(polyData, leafSize=0.01)
</pre>

But most fitting algorithms we have implemented don't attempt to protect themselves from this type of error.  The caller could also do the cropping.

patmarion, 2015-11-13:
Please re-open if you determine this is a bug in the code.  I think this looks like an issue with a particular dataset with extents that are too large.

closed 2015-11-13 by patmarion