RobotLocomotion/director #84: Build error on Mac 10.10 using vtk5 from macports

patmarion, 2015-10-07:
User posts build errors after installing vtk5 from macports.  Install command was:

port install vtk5 +qt4_mac qt4-mac python27 py-pip py-numpy

Build error is at link time:

[ 53%] Performing update step for 'ctkPythonConsole'
[ 56%] Performing configure step for 'ctkPythonConsole'
loading initial cache file /project/robotics/library/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/ctkPythonConsole-cache-Release.cmake
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   ctkPythonConsole

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /project/robotics/library/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/ctkPythonConsole-build
[ 59%] Performing build step for 'ctkPythonConsole'
[  7%] Linking CXX shared library ../lib/libctkPythonConsole.dylib
[ 85%] Built target ctkPythonConsole
[ 92%] Linking CXX executable ../bin/console
[100%] Built target console
[ 62%] Performing install step for 'ctkPythonConsole'
[ 85%] Built target ctkPythonConsole
[100%] Built target console
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /project/robotics/library/drake-distro/install/include/ctkPythonConsole/ctkAbstractPythonManager.h
-- Up-to-date: /project/robotics/library/drake-distro/install/include/ctkPythonConsole/ctkConsole.h
-- Up-to-date: /project/robotics/library/drake-distro/install/include/ctkPythonConsole/ctkPythonConsole.h
-- Up-to-date: /project/robotics/library/drake-distro/install/include/ctkPythonConsole/ctkPythonConsoleExport.h
-- Installing: /project/robotics/library/drake-distro/install/lib/libctkPythonConsole.1.0.dylib
-- Up-to-date: /project/robotics/library/drake-distro/install/lib/libctkPythonConsole.dylib
[ 65%] Completed 'ctkPythonConsole'
[ 75%] Built target ctkPythonConsole
[ 78%] Performing configure step for 'ddapp'
loading initial cache file /project/robotics/library/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/ddapp-cache-Release.cmake
-- sphinx-build not found - python API documentation will not be built.
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   _transformations
   ddApp
   vtkDRCFilters

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /project/robotics/library/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build
[ 81%] Performing build step for 'ddapp'
[  1%] Linking CXX shared library ../../../lib/libvtkDRCFilters.dylib
Undefined symbols for architecture x86_64:
  "vtkObjectBase::PrintHeader(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent)", referenced from:
      vtable for vtkGridSource in vtkGridSource.cxx.o
      vtable for vtkFrameWidget in vtkFrameWidget.cxx.o
      vtable for vtkFrameWidgetRepresentation in vtkFrameWidgetRepresentation.cxx.o
      vtable for vtkInteractorStyleTerrain2 in vtkInteractorStyleTerrain2.cxx.o
      vtable for vtkDepthImageProcessingPass in vtkDepthImageProcessingPass.cxx.o
      vtable for vtkEDLShading in vtkEDLShading.cxx.o
[... 38 lines cut ...]
  "operator<<(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent const&)", referenced from:
      vtkGridSource::PrintSelf(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent) in vtkGridSource.cxx.o
      vtkFrameWidgetRepresentation::PrintSelf(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent) in vtkFrameWidgetRepresentation.cxx.o
      vtkInteractorStyleTerrain2::PrintSelf(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent) in vtkInteractorStyleTerrain2.cxx.o
      vtkDepthImageProcessingPass::PrintSelf(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent) in vtkDepthImageProcessingPass.cxx.o
      vtkEDLShading::PrintSelf(std::basic_ostream<char, std::char_traits<char> >&, vtkIndent) in vtkEDLShading.cxx.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/build.make:383: recipe for target 'lib/libvtkDRCFilters.dylib' failed
gmake[10]: *** [lib/libvtkDRCFilters.dylib] Error 1
CMakeFiles/Makefile2:1044: recipe for target 'src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all' failed
gmake[9]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
gmake[8]: *** [all] Error 2
CMakeFiles/ddapp.dir/build.make:114: recipe for target 'src/ddapp-stamp/ddapp-build' failed
gmake[7]: *** [src/ddapp-stamp/ddapp-build] Error 2
CMakeFiles/Makefile2:180: recipe for target 'CMakeFiles/ddapp.dir/all' failed
gmake[6]: *** [CMakeFiles/ddapp.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
gmake[5]: *** [all] Error 2
Makefile:26: recipe for target 'all' failed
gmake[4]: *** [all] Error 2
CMakeFiles/director.dir/build.make:115: recipe for target 'director-prefix/src/director-stamp/director-build' failed
gmake[3]: *** [director-prefix/src/director-stamp/director-build] Error 2
CMakeFiles/Makefile2:739: recipe for ta

edowson, 2015-10-07:
Hi Pat,
I came across the following link that describes a way to specifically configure vtk to build against libstdc++, instead of libc++ (which is default for MacPorts when using gcc-4.8.x).

http://stackoverflow.com/questions/21546810/macports-on-osx-10-9-compile-with-stdlib-libstdc

I'm going to try with the following option now:
sudo port install -s vtk5 --with-qt configure.cxx_stdlib="libstdc++";

I'll let you know how it goes, and if it works, will submit a pull request for the working install_prereqs.sh file.

edowson, 2015-10-07:
It looks like no matter what I do to configure the vtk5 build, MacPorts ignores it.

I've used vtk before, so I'll just build it from sources and set the appropriate environment variables so that Director can find them.

patmarion, 2015-10-07:
Why do you want to use libstdc++ on your compiled libraries? Since libc++ is the default now, it's possible you may run into other compatibility issues.  Maybe just let everything use libc++?

edowson, 2015-10-07:
The errors with vtk on the pcl forums appeared to be because of not using libstdc++. But you're right, mixing this will definitely cause more issue. I'll try to build a standalone version of vtk-5.10.1 and see if it works, and then try to check the linked libraries.

edowson, 2015-10-07:
Hi Pat, 
Would you happen to know how to get past this cmake error? I'm getting it while attempting to configure vtk-5.10.1 sources. I have cmake-3.3.2 installed using macports.

CMake Error at /opt/local/share/cmake-3.3/Modules/TestBigEndian.cmake:51 (message):
   no suitable type found
 Call Stack (most recent call first):
   CMake/CMakeBackwardCompatibilityC.cmake:31 (TEST_BIG_ENDIAN)
   CMakeLists.txt:517 (INCLUDE)

patmarion, 2015-10-07:
I assume macports and homebrew patch vtk-5.10, or do something special to ensure configuration is correct, since Apple changes compilers with every OS update but vtk-5.10 source was released several years ago.  You might need a similar patch.

edowson, 2015-10-07:
I've temporarily forced TestBigEndian.cmake to return 0, taken from a patch here:
https://raw.githubusercontent.com/thewtex/emscripten/11630fb428db410ab93f6df2a1ff056485295667/cmake/Modules/TestBigEndian.cmake

I've replaced the contents of the file located in /opt/local/share/cmake-3.3/Modules/TestBigEndian.cmake with the above file.

It's now allowed the configuration to proceed.

edowson, 2015-10-07:
> 1) Could you post the output of:
>     otool -L /project/robotics/library/drake-distro/install/lib/libctkPythonConsole.dylib

```
$ otool -L /project/robotics/library/drake-distro/install/lib/libctkPythonConsole.dylib
/project/robotics/library/drake-distro/install/lib/libctkPythonConsole.dylib:
	/project/robotics/library/drake-distro/install/lib/libctkPythonConsole.1.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/local/libexec/qt4/Library/Frameworks/QtGui.framework/Versions/4/QtGui (compatibility version 4.8.0, current version 4.8.7)
	/opt/local/libexec/qt4/Library/Frameworks/QtCore.framework/Versions/4/QtCore (compatibility version 4.8.0, current version 4.8.7)
	/opt/local/Library/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.0)
	/project/robotics/library/drake-distro/install/lib/libPythonQt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libgcc/libstdc++.6.dylib (compatibility version 7.0.0, current version 7.21.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
	/opt/local/lib/libgcc/libgcc_s.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

> 2) Post the contents of drake/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build/CMakeCache.txt

```
# This is the CMakeCache file.
# For build in directory: /project/robotics/library/drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/ddapp-build
# It was generated by CMake: /opt/local/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Initial cache
BUILD_DOCUMENTATION:BOOL=OFF

//Build drake-designer with shared libraries.
BUILD_SHARED_LIBS:BOOL=ON

//Build the testing tree.
BUILD_TESTING:BOOL=ON

//Path to a program.
BZRCOMMAND:FILEPATH=BZRCOMMAND-NOTFOUND

//Path to a program.
CMAKE_AR:FILEPATH=/opt/local/bin/ar

//Initial cache
CMAKE_BUILD_TYPE:STRING=Release

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/opt/local/bin/c++

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release builds for minimum
// size.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during release builds.
[... 1649 lines cut ...]
bot2-lcmgl-renderer_pkgconfig_LIBS_L:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_LIBS_OTHER:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_LIBS_PATHS:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_PREFIX:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_STATIC_CFLAGS:INTERNAL=-D_REENTRANT;-I/project/robotics/library/drake-distro/install/include;-I/opt/local/include/glib-2.0;-I/opt/local/lib/glib-2.0/include;-I/opt/local/include
bot2-lcmgl-renderer_pkgconfig_STATIC_CFLAGS_I:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_STATIC_CFLAGS_OTHER:INTERNAL=-D_REENTRANT
bot2-lcmgl-renderer_pkgconfig_STATIC_INCLUDE_DIRS:INTERNAL=/project/robotics/library/drake-distro/install/include;/opt/local/include/glib-2.0;/opt/local/lib/glib-2.0/include;/opt/local/include
bot2-lcmgl-renderer_pkgconfig_STATIC_LDFLAGS:INTERNAL=-L/project/robotics/library/drake-distro/install/lib;-L/opt/local/lib;-lbot2-lcmgl-renderer;-llcmtypes_bot2-lcmgl;-llcm;-lgthread-2.0;-lglib-2.0;-lintl;-liconv;-Wl,-framework,Carbon;-Wl,-framework,Foundation
bot2-lcmgl-renderer_pkgconfig_STATIC_LDFLAGS_OTHER:INTERNAL=-Wl,-framework,Carbon;-Wl,-framework,Foundation
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBDIR:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBRARIES:INTERNAL=bot2-lcmgl-renderer;lcmtypes_bot2-lcmgl;lcm;gthread-2.0;glib-2.0;intl;iconv
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBRARY_DIRS:INTERNAL=/project/robotics/library/drake-distro/install/lib;/opt/local/lib
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBS:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBS_L:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBS_OTHER:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_STATIC_LIBS_PATHS:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_VERSION:INTERNAL=0.0.1
bot2-lcmgl-renderer_pkgconfig_bot2-lcmgl-renderer_INCLUDEDIR:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_bot2-lcmgl-renderer_LIBDIR:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_bot2-lcmgl-renderer_PREFIX:INTERNAL=
bot2-lcmgl-renderer_pkgconfig_bot2-lcmgl-renderer_VERSION:INTERNAL=
prefix_result:INTERNAL=/project/robotics/library/drake-distro/install/lib

```

patmarion, 2015-10-07:
Thanks.  So I can see that on libctkPythonConsole.dylib (which is just a library compiled along with Director) there are these libs:

    /opt/local/lib/libgcc/libstdc++.6.dylib
    /opt/local/lib/libgcc/libgcc_s.1.dylib

Looks like those are from a macports installed version of gcc.  I think you said you did something to your environment when compiling drake to force gcc.  I wouldn't do that.  I would just let drake compile with the default system compilers if possible (that would be xcode's clang).  These are the only compilers that are tested and supported.  In the CMakeCache.txt you posted I can see the project was configured with this c++ compiler:

/opt/local/bin/c++

edowson, 2015-10-13:
Based on feedback from pierluigi, the following patch ought to fix the build issue.

'''
diff --git a/distro/pods/drake-distro/install_prereqs.sh b/distro/pods/drake-distro/install_prereqs.sh
index 0b069fc..e599ab7 100755
--- a/distro/pods/drake-distro/install_prereqs.sh
+++ b/distro/pods/drake-distro/install_prereqs.sh
@@ -8,7 +8,9 @@ case $1 in
     brew install python numpy
     brew install vtk5 --with-qt ;;
   ("macports")
-    echo "WARNING: install_prereqs macports not implemented for this module" ;;
+    port install qt4-mac qwt
+    port install python27 py-pip py-numpy py-pyqt4
+    port install vtk5 +qt4_mac +python27 ;;
   ("ubuntu")
     apt-get install libqt4-dev libvtk5-dev libvtk5-qt4-dev python-vtk python-numpy ;;
   ("cygwin")

'''

pierluigiM, 2015-10-13:
The first post is about the error "QVTKWidet.h" not found. The point here is in installing VTK5 with Qt via macports. I did not touch compiler settings, and downloaded VTK5 with the following string:
`sudo port install vtk5 +qt_4mac +python27`
This should create the right Qt compatible files for VTK5. Of corse I had downloaded cmake.

patmarion, 2015-10-13:
Thanks for the updates.

edowson, 2015-10-13:
I've submitted the following patches to director, iris-bistro and signal-scope to update the install_prereqs.sh scripts for MacPorts:

edowson/iris-distro@6babfc0b09b1795ed7dd79726e585640303b7458

edowson/director@264683252ee7129f30f1322e1d90949f375e8544

edowson/signal-scope@4c2cdb02b137d8c0305f4cb55613712105c1e387

patmarion, 2015-10-13:
Thanks.  I've opened a pull request in drake to bump up the director and signal-scope externals to include your patches:

https://github.com/RobotLocomotion/drake/pull/1424

pierluigiM, 2015-10-13:
edowson, have you tried "make test". If you did, any failed one?

edowson, 2015-10-14:
@pierluigiM There were 16 tests that failed out of 482 (that was the total number, I think). Around 96% of the tests passed. Some of them were related to Atlas, and from the looks of it, failed due to the accuracy or the results not being within tollerance limits. It might be related to the backend solvers that were used with my test configuration. I used WITH_GUROBI=ON, and set WITH_ALL_PUBLIC_EXTERNALS=OFF.