RobotLocomotion/director #91: drake python include fails on drakevisualizer.py

mauricefallon, 2015-10-19:
I don't think the fix in this commit works:
ce35723e54707787c5e31872231abefd93cf918f

the problem is that drake python bindings are now installed to:
dist-packages/lcmtypes/drake
Previously it was to:
dist-packages/drake

It finds the package but cannot import drake types included within the first imported file.

  File "/home/mfallon/ipab-distro/drc/software/build/lib/python2.7/dist-packages/lcmtypes/drake/__init__.py", line 12, in <module>
    from .lcmt_qp_controller_input import lcmt_qp_controller_input
  File "/home/mfallon/ipab-distro/drc/software/build/lib/python2.7/dist-packages/lcmtypes/drake/lcmt_qp_controller_input.py", line 12, in <module>
    import drake.lcmt_joint_pd_override
ImportError: No module named drake.lcmt_joint_pd_override

This is because drake doesn't install its python lcmtypes in a conventional locaiton

patmarion, 2015-10-19:
I reported it here:

https://github.com/RobotLocomotion/drake/issues/1425

Unfortunately, it's a appears to be an issue in lcm, exposed by drake's non-default install directory.  A couple options to fix:

submit a bugifx to lcm and then in the meantime:
 - require an older version of LCM for DRC
 - require a patched version of LCM for DRC
 - revert the drake install location (but then we need to re-open @rdeits motivation for changing the install location)
  - use sed or something to fix the generated .py files

mauricefallon, 2015-10-19:
Am I right that the reason to rename it from 'drake' was because of the wish to use the drake namspace in python for drake bindings - not the lcmtypes?

Your 3rd suggestion seems straightforward but what about installing the python bindings to a different folder e.g. lcmdrake or drakelcm?

Options 1 and 2 would be troublesome

patmarion, 2015-10-19:
I think the trouble with installing to lcmdrake or something like that is that it requires changing the lcm package name, which affects matlab, java, and c code files.  For Python, it was simple to just install the drake package in a subdirectory and update the import statement.  This should work, but the lcm bug in lcm-gen wasn't detected when Robin rolled out this change, because Drake uses lcm-gen 1.0.

@rdets what do you think about option 4, to use sed or something to patch the genreate .py files, basically replace "import drake." with "import lcmtypes.drake."  Also, I don't know if this would be tricky, but really a more logical import statement might be import drake.lcmtypes :)

rdeits, 2015-10-19:
Yeah, option 4 is doable. But I'd really like to get this fixed in LCM itself, since it's a pretty obvious bug. Have we submitted that bug report upstream yet?

rdeits, 2016-01-26:
I've given up on using the `drake` namespace for the main python bindings, so we can go back to using the default LCM install package name `drake`. This should clear up the issues we're having with newer versions of LCM.

closed 2016-08-09 by wxmerkt