RobotLocomotion/director #95: odd logic in teleoppannel.py

mauricefallon, 2015-10-22:
When debugging https://github.com/RobotLocomotion/director/pull/90 I noticed that this doesn't do anything:

          if self.panel.ikPlanner.defaultIkParameters.useCollision:
            self.setLHandConstraint('ee fixed')
          else:
            self.setLHandConstraint('ee fixed')
        elif side == 'right':
          if self.panel.ikPlanner.defaultIkParameters.useCollision:
            self.setRHandConstraint('ee fixed')
          else:
            self.setRHandConstraint('ee fixed')

...In teleoppanel.py

Any idea what the intention was? Andres added is some time ago

patmarion, 2015-10-22:
I think we can remove it now.  I think when Andres was testing he would set the constraint to something other than 'ee fixed' when collision was enabled (He'd set to 'orbit', which is a fancier reach constraint that only works for certain affordances).  But before merging into master he set it back to 'ee fixed', so it's just extra logic that doesn't do anything.

avalenzu, 2015-10-22:
That sounds about right, Pat. Sorry for the mysterious, do-nothing code!