RobotLocomotion/director #132: mesh scaling

mauricefallon, 2015-11-27:
I have a urdf with mesh files in mm. The 0.001 scale factor in the following tag isn't respected in the director (which uses the drake parser):

    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry name="Waist_visual">
        <mesh filename="package://bot_model/meshes/Waist.obj" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="grey"/>
    </visual>

Any idea where should I add this scaling? Happy to dig around for it myself

FWIW, I noticed that the scale tag is sometimes present in the Atlas model but presumably ignored as its just unity:

    <link name="right_convex_palm">
    <collision group="right_convex_hand_robotiq">
      <origin rpy="-1.57079 -1.57079 0" xyz="0 0.1725 0"/>
      <geometry>
        <mesh filename="package://robotiq_/meshes/GRIPPER_OPEN_chull.obj" scale="1 1 1"/>
      </geometry>
    </collision>
    </link>

patmarion, 2015-11-27:
I fixed this in drake already, but perhaps the open humanoids version of
drake does not have the fix?  We could cherry pick it over.  I'm not at my
computer now to look it up, it was a mod to systems/plants/shapes

patmarion, 2015-11-27:
should be fixed now.  I had an old patch lying around that was required for director, but it hadn't been applied back then because drake/master needed to be updated.

closed 2015-11-27 by patmarion