RobotLocomotion/director #140: VTK build on OSX is very slow

rdeits, 2015-12-16:
(too slow for Travis, for example). 

We should look into providing a bottled version of vtk5 with python.

rdeits, 2015-12-16:
It looks like Homebrew doesn't support bottles for optional arguments: https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/FAQ.md#why-do-you-compile-everything

Perhaps we could package our own precompiled VTK? Or even create a vtk5-python recipe and bottle that?

RussTedrake, 2016-01-23:
what do you guys think about packaging director as precompiled vs vtk5 as precompiled?

patmarion, 2016-01-23:
I think the issue of being too slow for travis is resolved by having a build script that prints progress, travis assumes your build has stalled if it goes for too long without printing progress.

is the main concern that the build is too slow for new users, or too slow for build servers?  do you image the precompiled option being a non-default option that is enabled only for build servers, or a default option enabled for users?

if we eventually enable features in drake-visualizer that link against drake (as we currently do in director), then that dependency must be compiled from source.

RussTedrake, 2016-01-23:
I will let @rdeits confirm, but I think this came in when students in 6.832 were turned off by the incredible build time.

patmarion, 2016-01-23:
I don't know why homebrew doesn't provided a bottled version of vtk, but it might have to do with the +python option.  On mac, it can be unpredictable whether the user wants system python, homebrew python, or something else (like anaconda python).

rdeits, 2016-01-23:
Yeah, the issue is the `--with-python` argument. And I think both issues came up: students were annoyed by the very long build time, and travis was stalling (and it wasn't clear how to ensure that homebrew printed enough output to keep it happy).

patmarion, 2016-02-04:
An update: a found a trick to get travis to homebrew install vtk without timeout (it takes over 25 minutes).  Then I had the travis server make a homebrew bottle for vtk and upload it.  In future builds, the bottle can be installed quickly.  However, it didn't work for macosx gcc builds.  I could make a gcc specific bottle.  The homebrew server runs osx mavericks, so it probably won't work for systems earlier than that.

Command to install bottle: https://github.com/RobotLocomotion/drake/pull/1689/files
Failure on osx with gcc: https://travis-ci.org/RobotLocomotion/drake/jobs/106893992

jhoare, 2016-03-18:
conda has vtk 5.10 and 6.3 packages. http://conda.pydata.org/docs/

I'm not sure how open you guys would be to using a non-system python though. If you're not familiar with conda, the short story is its a python distribution that has binary packages for a lot of existing python packages. We often use it to more easily get a more recent python version on older systems. They provide packages for linux, windows, and osx.

patmarion, 2016-03-18:
thanks for the info.  We already use a non-system python when doing homebrew installs (although it also works with system python if the user chooses to avoid homebrew python)