RobotLocomotion/director #171: Required lcmtypes for visualisation of robot, sensors and plans

mauricefallon, 2016-02-02:
To allow the director to minimally operate without the full OH stack, what are the  lcmtypes needed to:
- draw the live robot state
- render some data. For example I think libbot can skipped if the point cloud accumulation is done e.g. by maps.
- make and execute plans

(this is with a view towards making an lcmtype repo independent of Drake, pronto, director and OH.)

mauricefallon, 2016-02-22:
@patmarion : you might have seen I started this:
https://github.com/openhumanoids/bot_core_lcmtypes
It currently contains only the libbot lcmtypes. I want to move most major drc_lcmtypes into it:
- states, lidar, images, planning messages, pointclouds

Question: what set of lcmtypes are needed to achieve what i said previously above?

(Another future goal would be to split up libbot and retire bot2-lcmgl and bot2-vis. And to switch to the maintained version of [procman](https://github.com/ashuang/procman)

patmarion, 2016-02-22:
draw live robot state: robot_state_t

load robot model: robot_urdf_t

manip plans:  robot_plan_w_keyframes_t.  But director doesn't actually use the keyframes, the keyframes are empty, so the main dependency is a robot_plan_t.  robot_plan_t has extra fields that we don't use in director, such as matlab_data and grasp transitions.  Also, robot_plan_with_supports_t is used.

footstep plans: footstep_plan_t and associated messages

Director has lots of lcmtype dependencies when it executes startup.py, for example irobot and robotiq hands, multisense, maps data requests, iris planner, etc.

I created a minimal app to draw robot state, manip plans, pointcloud_t, and image_t.  The app imports much more limited set of modules.  Those modules import the drc lcmtypes, but this might be a good starting point to edit  those modules to limit their lcmtypes to just the bot_core_lcmtypes.  

https://github.com/RobotLocomotion/director/pull/199

You can run the minimal app with Atlas v5 and DRC finals log:

directorPython testDrawRobotLog.py -v5

mauricefallon, 2016-02-22:
That's awesome, thanks! I'll work with this and add a demo with a pointcloud_t stream.
Yes, the plan messages could do with being redesigned - they have a lot of legacy stuff in there.