RobotLocomotion/director #192: Screen Grabber returns unexpected error message

vtjeng, 2016-02-12:
When I click the "Save Screenshot" button on the Screen Grabber Panel generated by `director.screengrabberpanel.ScreenGrabberPanel(view)`, the following error message appears.

```
Traceback (most recent call last):
  File "/home/vtjeng/director/build/install/lib/python2.7/dist-packages/director/screengrabberpanel.py", line 124, in onSaveScreenshot
    app.getMainWindow().statusBar().showMessage('Saved: ' + filename, 2000)
AttributeError: 'NoneType' object has no attribute 'statusBar'
```

This seems to be a reproducible error.

patmarion, 2016-02-13:
Thanks for the report.  I think it still saves screenshots, despite the error?

vtjeng, 2016-02-13:
Yes, I can confirm that it still saves screenshots.

patmarion, 2016-03-10:
this has been fixed.

closed 2016-03-10 by patmarion