RobotLocomotion/director #202: Add remote rendering types to director?

mauricefallon, 2016-02-24:
Pre-DRC I heavily used a libbot renderer called "LCM collections". It allows a process to send basic drawing primitives over LCM to be rendered. We used it a lot for SLAM and algorithm visualisation.

It was quite similar to the concept in ROS of Markers - see this video:
http://wiki.ros.org/rviz/DisplayTypes/Marker
http://docs.ros.org/api/visualization_msgs/html/msg/Marker.html

Implementing this would seem to be easy in VTK. I think that Drake could really do with this as LCM GL is hard to use.

@patmarion @RussTedrake : what do you think?

patmarion, 2016-02-24:
Agreed, that would be a good addition.  The drake-visualizer supports some of this already, but it is robot and link oriented.  @mauricefallon would you recommend the collections types, or a clone of the markers messages?

mauricefallon, 2017-01-17:
this has been integrated

closed 2017-01-17 by mauricefallon