RobotLocomotion/director #210: Add Joint Frame Visualization?

liangfok, 2016-03-07:
Can joint frames be visualized in Director / Drake Visualizer? Here's an example of what it looks like in Gazebo:

[image]

patmarion, 2016-03-07:
No, not currently.  The visualizer doesn't receive kinematic tree information from drake, so it does not have access to this information.  If we want to add this feature, we could open a new issue under drake to expand the rigid body tree drawing commands to provide the visualizer with the kinematic information.

The visualizer does draw the link frame for each body, which is often in line with the joint frame.  You can open the object model (press F1 or use the View menu) and then expand the tree to explore the bodies and frames.

liangfok, 2016-03-07:
Thanks for the quick reply. The ability to visualize joint frames is sufficient for now.

closed 2016-03-07 by liangfok

patmarion, 2016-03-07:
By the way, you can right-click on a body in the 3D view to open a context menu.  From the menu, pick "select" and that will expand the object model automatically to the selected object.  You can also double click an object to show/activate its frame.  Double click again to deactivate.