RobotLocomotion/director #216: Relative to Director ui updates

simalpha, 2016-03-10:
I refer to the Footstep Panel in Director. I would like to re-organize the group of buttons contained in the QGroupBox named "Advanced Options". In particular:

1. add the two properties "IHMC Transfer Time" and "IHMC Swing Time" to the list
2. split the list in 2 groups: "Planning Options" and "Control Options"
3. split the "Control Options" group in subgroups: "Drake" and "IHMC"

Now, I can create many QGroupBox in the file ddFootsteps.ui, but then I have to populate them from footstepdriverpanel.py. Currently, the constructor of the FootstepsPanel class populates the "Advanced Options" box with the whole list of properties from the footstepDriver params (`self.panelConnector = propertyset.PropertyPanelConnector(self.driver.params.properties, self.propertiesPanel)`), without any distinction. I am not sure about how to add planning and control options separately in footstepdriver.py, in order to be able to refer to different lists from the footstepPanel.

@patmarion Any suggestion? What would you do?  

Thanks!

patmarion, 2016-03-10:
The parameters in driver.params.properties are copied to fields of the lcm message types footstep_plan_params_t and footstep_params_t.  Do the IHMC parameters (transfer time, swing time) already exist in the footstep_params_t message type?

Note, so far have have drake_* and bdi_* parameters.  So there is precedent for having different controllers, drake and bdi, and now ihmc.  But we have listed all the parameters in the same container (driver.params) and certain bdi parameters are not exposed, instead they are hard coded in FootstepsDriver.getDefaultStepParams().

I think your suggestion of splitting the parameters into new groups sounds like a good one.  It would start with splitting up the driver.params into multiple property set objects.  But first, I'm curious about the lcm messages.  Do we want to continue putting everything into footstep_params_t, or start introducing controller specific message types?

simalpha, 2016-03-10:
The IHMC parameters do not exist in the footstep_params_t message type. I initially thought about adding them to the message, but, as you pointed out, it already contains BDI-specific parameters (bdi_step_duration, bdi_sway_duration) and Drake-specific parameters (step_speed). Instead, since we only need two params for IHMC, what about publishing a  double_array_t message to a new IHMC_FOOTSTEP_PARAMS channels?

patmarion, 2016-03-10:
Calling  @rdeits into the discussion :)

rdeits, 2016-03-10:
Yup, reorganizing the parameters into planning and control sounds great. I'm all for it :)

I'm pretty firmly against the idea of sending a second message with additional params, though, since it introduces a whole new world of opportunities for bugs (what if we miss one message but not the other? what if we read a footstep_params_t that doesn't correspond to the last double_array_t that we got? etc...). 

I would propose creating a new lcmtype for footstep plan requests that contains a `string controller_params` field, which we can use to hold a json-encoded set of parameters for whatever controller has been chosen.

mauricefallon, 2016-03-10:
BTW, the IHMC parameters are not used in any planning. they are only appended to footstep positions from a typical footstep plan when sending the plan to IHMC's controller. We do this in our ROS-to-LCM translator by intercepting the WALKING_PLAN_REQUEST message.

So the IHMC parameters are not in anyway connected to the current footstep_params_t. I believe we should avoid conflating them with what goes to Robin's planner.