RobotLocomotion/director #237: Getting head context for valkyrie

mauricefallon, 2016-03-28:
In segmentationroutines.py there are methods for getting an abstract context meaning "the direction the robot is looking". 

https://github.com/RobotLocomotion/director/blob/master/src/python/director/segmentationroutines.py#L109

Unfortunately the "head" frame of Valkyrie is inverted relative to Atlas (z-axis is down in the default posture). The minimal solution is add a frame to the URDF that is the same as head - but rotated by 180.

And then having this function `self.model.getHeadLink()` return "head_rotated (for example)

edbot, 2016-03-29:
I will resolve this issue for now by using upperNeckPitchLink for this in the director config and see hw it works out. It has the correct orientation (-/+ a few degrees) and is rigidly attached to "head"

(maurice) 

[image]

patmarion, 2016-03-29:
Was this causing a bug someplace?  I can find several places where the view frame is accessed, but the only function I can find that accesses the z axis of the view frame is the function addCoordArraysToPolyData().

All the other functions use the view frame in order to extract the view origin and view direction.  We could update all those places to call the context.getViewOrigin() and context.getViewDirection().

Perhaps we should explicitly define the view frame, or remove the function context.getViewFrame() so that we don't write code that depends on an non standard definition of view frame.