RobotLocomotion/director #273: Building drake-visualizer without Drake

rdeits, 2016-05-28:
I've been using the drake-visualizer app a lot over the past few weeks from outside the Drake ecosystem (specifically, from https://github.com/rdeits/DrakeVisualizer.jl ), and I think other people might find that to be very useful as well. I think it would be nice to be able to build drake-visualizer without installing Drake itself. As far as I can tell, the only thing stopping us from doing that is the fact that the LCM type definitions for `lcmt_viewer_load_robot` and friends live in Drake. Is that correct? 

If so, I think it would be reasonable to duplicate the message types here so we can build the whole app without Drake.

I'm happy to do the implementation myself. What do y'all think?

liangfok, 2016-05-28:
How about pulling the LCM message definitions into a separate repository / library that both Drake and Drake Visualizer depend on? That would ensure the messages used by Drake and Drake Visualizer remain in sync.

rdeits, 2016-05-29:
Yeah, I considered that too. But I'm not sure it's actually any easier, since changing the types would still require changes in both Drake and Director to ensure they point to the same revision of that new library. Of course, there's value in preventing duplication, but these definitions are pretty small (about 60 lines total), so I'm not sure it's worth adding yet another repo for us to manage. What do you think?

RussTedrake, 2016-05-29:
I vote no on the additional repo.  If director is the only visualizer we support now that is using this message type, i would rather consider moving that lcm-type into director (and out of drake).

liangfok, 2016-05-29:
+1 for moving the LCM types into Director.

rdeits, 2016-05-29:
Agreed, that sounds good. I'll give it a try.

RussTedrake, 2016-05-29:
Hold on.  It was an idea, but we need to think it through.  Right now, drake builds a lot of libraries / examples that use botvisualizer.  If it builds without director, then those can still compile -- but simply send their messages to the ether (or to a precompiled version of director).  If we execute this plan, then we would have to disable all of the botvisualizer examples if we're building drake w/o building director.  That sounds pretty bad -- so i'd want to decide on an alternative before going down this road.

liangfok, 2016-05-29:
I'm not sure I understand the problem. If Director is disabled in Drake and the Drake-Director LCM messages are defined in Director, it makes sense to also disable all libraries and examples in Drake that use Director's LCM messages. I'm not sure how these libraries and examples can still compile if the definitions of the LCM messages that they use are no longer present. Am I missing something?

patmarion, 2016-05-29:
Hi guys!  As Russ points out, if you put lcm types into director then c++ programs that wish to use the lcm types, such as BotVisualizer.h, have to build director as a library dependency.  I've so far resisted adding any lcm types to director (it does not export any, nor have any CMake targets setup to run lcm-gen etc).  Maybe bot-core-lcmtypes is a reasonable location that is already used by both drake and director?

avalenzu, 2016-05-29:
Wouldn't moving these LCM types into director introduce a circular
dependency when building Drake and director with USE_DRAKE=true? Or can
CMake figure out that it needs to build some libraries from Drake, then
some from director, then more from Drake?

rdeits, 2016-05-30:
@patmarion bot_core_lcmtypes seems like a great option, which would also prevent any circular dependencies

rdeits, 2016-05-31:
Actually, it looks like Director has a dependency on `RobotLocomotion/libbot`which includes `bot_core` and some lcmtypes in `bot_core/lcmtypes`, but it doesn't yet depend on the standalone `bot_core_lcmtypes` repo which contains more core types. @patmarion is it OK to add that dependency?

patmarion, 2016-05-31:
Sounds good!

rdeits, 2016-06-06:
Ok, I have this working in #275

patmarion, 2016-10-26:
This is resolved now, the draw messages are provided by bot_core_lcmtypes (and robotlocomotion/lcmtypes)

closed 2016-10-26 by patmarion