RobotLocomotion/director #285: collectionsmanager fails on startup

manuelli, 2016-06-27:
@mauricefallon Currently launching director fails on startup with 

`  File "<string>", line 7, in <module>
  File "/home/manuelli/drc/software/build/lib/python2.7/dist-packages/director/startup.py", line 305, in <module>
    app.MenuActionToggleHelper('Tools', 'Renderer - Collections', collectionsManager.isEnabled, collectionsManager.setEnabled)
AttributeError: 'NoneType' object has no attribute 'isEnabled'`

on my system I don't have vtkCollections apparently and so lcmcollections.init(view) returns a `None` object. In any case director shouldn't fail on startup like this. It's as easy as adding a check on whether collectionsManager is None.

Is vtkCollections something that should be part of the vtkAll package in director? Did I mess something up in the build to cause this?

mauricefallon, 2016-07-02:
I added this last week. Do you have success with a `make clean; make` in the director path?
it resolved build issues for a few others here.

mauricefallon, 2016-10-05:
this is resolved with a clean make. we've been using it for sometime

closed 2016-10-05 by mauricefallon