RobotLocomotion/director #311: Update Superbuild to Not Default to USE_SYSTEM_VTK=ON

liangfok, 2016-10-09:
Instead, the superbuild should check if `vtk5` is installed on the system. If it is not, it use use `USE_SYSTEM_VTK=OFF`.

This was originally mentioned here: https://github.com/RobotLocomotion/drake/issues/2624#issuecomment-251938584

liangfok, 2016-10-09:
@patmarion, I'd like to assign this ticket to myself but can't. Can I get the bits?

patmarion, 2016-10-09:
Even though I made the comment I'm not sure I want the feature.  I don't always like top level options that automatically set based on the environment.  I usually prefer default options to be predictable.

But there should definitely be a mechanism in place to catch the issue at configure time.  In your case, director found vtk6 and succeeded to configure, but it should have been an error.  If vtk isn't installed at all, then the issue will be caught at configure time, im fine with that.

liangfok, 2016-10-09:
OK I'll look into updating the super-build system to by default verify at configure time whether `vtk5` is installed on the system, and to error out if it is not.

patmarion, 2016-10-24:
I'll close this one now.  Sounds like the logic will be added to drake's superbuild, not director's.

PR #339 (2016-10-24): Added logic to fail fast if user wants to use system VTK but version 5 is not available.
I tested on this on Ubuntu 16.04 with ROS Kinetic (which has VTK6) and it worked -- the error was printed during the configure stage. I also tested it on Ubuntu 14.04 with ROS Indigo and Director was able to compile using the system VTK.

Resolves #311.
closed 2016-10-24 by patmarion