RobotLocomotion/director #320: Consolidate IK front end in director

patmarion, 2016-10-12:
There is the plannerPublisher and ikServer.  There are just a few branches in the code:

In JointLimitChecker:
<pre>
    # update limits on server
    panel.ikPlanner.ikServer.updateJointLimits(limitData)
    panel.ikPlanner.plannerPub.updateJointLimits(limitData)
</pre>

In ikplanner.py:
<pre>

        if self.planningMode == 'drake' and self.pushToMatlab:
            self.ikServer.sendPoseToServer(pose, poseName)
        elif self.planningMode == 'exotica':
            self.plannerPub.processAddPose(pose, poseName)

    # and two more branches in runIk() and runIkTraj()
</pre>

I am now adding support for a pydrake ik server.  I know exotica also supports a drake mode.  The mode I am adding will allow director+drake to provide IK applications without the full oh-distro stack.

My plan is:

Remove the branches.  The api to plannerPub and ikServer are nearly identical.  I will keep the plannerPub api as is and remove the ikServer.  I will make PlannerPublisher a base class that defines the api, then derive subclasses to support: exotica, pydrake, and matlab drake.

All this will be minimal code changes, mostly just tweaking the PlannerPublisher to separate the lcm publishing from the constraint processing.  I just wanted to make this post so that people are aware of my plans.  I'll have a PR open later this week.

If I get to it, I also want to make IK planning asynchronous so that the Director UI does not freeze while waiting for long planning problems, but this might take some extra effort.

mauricefallon, 2016-10-12:
Perfect. We (Vlad, Yiming mostly) had discussed a plan for this but hadn't acted on it. What you are describing closely matches what we wanted to do. Eventually we would like to move to use exotica over python.

This is how Exotica solving an IK problem looks:
https://youtu.be/e09B-hAVKNs?t=20
The UI input comes from Director and streams constraints over LCM to Exotica which replies with the IK solution. (Exotica is using Drake-C++ internally). On an I-O level it behaves just like IK with matlab-drake. 

It's like we have a Boston office to out source things we can't do ourselves!

sammy-tri, 2016-10-12:
+1  (of course, I would approve of this, since it's what I was aiming to do with my half-baked pydrake implementation)

patmarion, 2016-10-17:
Ok this is resolved (mostly) in https://github.com/RobotLocomotion/director/pull/324 now.  This also introduces a pydrake back end which is about 85% complete.

We still have code in places that calls the matlab ikserver directly, instead of going through the general ik interface.  For example, a lot of the RRT and end pose search code.  That stuff is only implemented in matlab.  But that's fine!  It would be nice to have tests for it though.

patmarion, 2016-10-27:
remaining todos are tracked here: https://github.com/RobotLocomotion/director/pull/349

closed 2016-10-27 by patmarion