RobotLocomotion/director #321: Video encoding command-line hint could/should specify "-pix_fmt yuv420p"

maddog-tri, 2016-10-13:
At https://github.com/RobotLocomotion/director/blob/b22e2490eb3b710fdb24aca760b96b4eedd3eff5/src/python/director/screengrabberpanel.py#L212, the suggested command-line for video encoding would probably benefit from an additional argument of `-pix_fmt yuv420p`.

Many (most?) linux distributions have already, or soon will, switch their default video processing libraries from "libav" back to "ffmpeg" (there is apparently a huge backstory of forking and feuding). `avconv` is provided by libav, and (perhaps?) also by ffmpeg (via a symlink to `ffmpeg`, at least on Ubuntu 16.04). However, the default chroma subsampling format for ffmpeg has changed from yuv420p to yuv444 --- yet yuv420p is required by many "dumb players", including Quicktime.  So, to ensure that the suggested command-line continues to create videos that work everywhere, explicitly selecting yuv420p is prudent.

See https://trac.ffmpeg.org/wiki/Encode/H.264#Encodingfordumbplayers for reference.

patmarion, 2016-10-13:
Thanks!  I agree it's worth adding to the recommended command line.

patmarion, 2017-05-01:
This was resolved last year with d8d8008b