RobotLocomotion/director #343: Right Click -> Copy Pointcloud broken

wxmerkt, 2016-10-26:
Traceback:
```
Traceback (most recent call last):
  File "/home/wxm/dev/oh-distro-private/software/build/lib/python2.7/dist-packages/director/robotviewbehaviors.py", line 380, in onCopyPointCloud
    polyData = segmentation.addCoordArraysToPolyData(polyData)
  File "/home/wxm/dev/oh-distro-private/software/build/lib/python2.7/dist-packages/director/segmentation.py", line 529, in addCoordArraysToPolyData
    viewOrigin = viewFrame.TransformPoint([0.0, 0.0, 0.0])
AttributeError: 'NoneType' object has no attribute 'TransformPoint'
```

patmarion, 2016-10-26:
<pre>
    def getViewFrame(self):
        return self.model.getLinkFrame(self.model.getHeadLink())
</pre>

Is it possible the robot that is loaded does not define a head link in the director_config.json?

We could fix this so it makes a more helpful error.  Instead of letting model.getLinkFrame() return None, it should assert that the headLink exists before asking for the head line.

patmarion, 2016-10-26:
fyi, in roboturdf.py:

    def getHeadLink(self):
        return drcargs.getDirectorConfig().get('headLink')

commit 41d6eeb7 (2016-10-26): warn if headLink undefined, closes #343
closed 2016-10-26 by patmarion