RobotLocomotion/director #351: Make movies with povray

patmarion, 2016-10-27:
VTK has an exporter for povray.  It actually works.  With a little TLC this might be useful for making flashier movies.

[image]

rdeits, 2016-10-27:
This is super exciting! Having to play my trajectories back at 1 fps or lower to render movies in Director was getting annoying ;-)

patmarion, 2016-10-29:
A few test renders- 

http://people.csail.mit.edu/patmarion/software/director/media/movies/

wxmerkt, 2017-03-13:
Hi Pat,
Any idea on when this feature might land? I've seen your branch (https://github.com/RobotLocomotion/director/compare/master...patmarion:add-povray-exporter) and wondered where and how we can help with TLC to get it into master. We're facing the issue of having to play back at 1-fps as well.
Cheers,
Wolf

patmarion, 2017-03-13:
i don't have any plans to continue developing this branch in the near term, but I'd be happy to merge it into master after just a little clean up.  It's basically in a working state, except I'm sure I broke a couple things when I rebased it.

I can make the code in moveexporter.py general purpose and move all the custom code into an example script.  There are also some pov-ray files required that are not in this branch but I have them in my dropbox.

Maybe we can schedule a time to talk on gchat while we try to get it working on your machine, then you can take it from there?

wxmerkt, 2017-03-13:
Sounds great - any time after Thursday is good, perhaps early next week?