RobotLocomotion/director #353: ImportError: No module named lcm

blegat, 2016-10-28:
The visualizer gets the following failure when I try to run it:
```
$ make superbuild
...
$ ./build/install/bin/drake-visualizer 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/blegat/git/director/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 3, in <module>
    from director import lcmUtils
  File "/home/blegat/git/director/build/install/lib/python2.7/dist-packages/director/lcmUtils.py", line 1, in <module>
    import lcm
ImportError: No module named lcm
```

patmarion, 2016-10-28:
When you `make superbuild` it compiles with a minimal configuration, without lcm.  Here's how you can enable lcm on your existing build:

$ cd build
$ cmake -DUSE_LCM=ON .
$ make -j2

Or, if you're familiar with the ccmake tool, you can run that in the build dir and browse the other options, too.

blegat, 2016-10-28:
It works, thanks !
Isn't scrolling supposed to zoom in/zoom out ?

rdeits, 2016-10-28:
hold right-click and move the mouse up and down to zoom

blegat, 2016-10-28:
That makes sense, left-click to rotate, right-click to zoom, middle-click to move :)

closed 2016-10-28 by blegat