RobotLocomotion/director #363: Update header includes to use files in drake/multibody/

amcastro-tri, 2016-11-07:
PR RobotLocomotion/drake#4058  moves all the RBT related files into `drake/multibody`.
To avoid breaking `Director`, PR RobotLocomotion/drake#4058 creates the following dummy files:

- drake/systems/plants/RigidBodyTree.h
- drake/systems/plants/ForceTorqueMeasurement.h
- drake/systems/plants/shapes/Geometry.h

so that the non-updated version of Director can still work. These files merely include the files in `drake/multibody`.

Post RobotLocomotion/drake#4058 two PR's should be submitted:

1. PR in Director updating includes to use RBT related files in `drake/multibody`
2. PR in Drake removing the dummy files cited above.

patmarion, 2016-11-07:
I don't think it's necessary to stage things this way.  You can just develop your changes to director in a branch and your drake branch can reference it (instead of director/master).  Then when your changes are reviewed and ready to merge into drake, we merge the director PR, then update your drake branch to reference it.

amcastro-tri, 2016-11-07:
Thank you @patmarion, this seems like a better idea. I pushed #364. PTAL.

amcastro-tri, 2016-11-09:
After #364 Director now includes files from `drake/multibody` and thus this issue is closed.

closed 2016-11-09 by amcastro-tri