RobotLocomotion/director #378: Camera Control "track modes" no longer do tracking in drake-visualizer

maddog-tri, 2016-11-17:
Activating a "Track Mode" via the Camera Control Panel in drake-visualizer doesn't quite work any more. It seems like the state of target is grabbed at the moment that the target is selected, and never updated after that.

I am pretty positive that this regression is a side-effect of the changes in #362.

patmarion, 2016-11-18:
yes it is.  I'll make a fix tomorrow, thanks!

patmarion, 2016-11-18:
in the meantime, if its blocking, you can cd to externals/director and revert commit 1570241

mauricefallon, 2016-11-18:
+1 I had noticed that too

patmarion, 2016-11-18:
just to confirm, @mauricefallon this bug is specific to the drake-visualizer app, is that what you are using?  Or do you also see a bug in Director?

mauricefallon, 2016-11-18:
scratch that. just checked. nevermind

patmarion, 2016-11-18:
sorry @maddog-tri, didn't have a chance to get to it today.  I'll let you know.

maddog-tri, 2016-11-18:
Oh, no rush at all --- I'm not blocked on this (...until someone says "WE NEED A DEMO VIDEO FROM THAT CAR'S POV! RIGHT NOW!").

PR #383 (2016-11-23): if a child frame exists then use it
this code was recently modified to optimize the viewer for scenes with
many links.  but the camera follower needs to use child frames.

fixes #378.
commit d5634ad6 (2016-11-23): if a child frame exists then use it

this code was recently modified to optimize the viewer for scenes with
many links.  but the camera follower needs to use child frames.

fixes #378.
closed 2016-11-23 by patmarion