RobotLocomotion/director #389: Drake Visualizer with multiple robots

rdeits, 2016-11-29:
I've run into an issue that's pretty hard to get around with the way `drake-visualizer` is currently set up. Specifically, I would like to:

* load robot A in drake visualizer once
* at each iteration of my function:
  * draw robot A in a configuration
  * load and draw another robot B (whose geometry changes at every iteration)

I can't currently do that, because loading robot B deletes all robots from the visualizer. I could do:

* at each iteration:
  * load robots A and B
  * draw robot A
  * draw robot B

but it's computationally wasteful, since it loads robot A every time (which might be expensive, if robot A has fancy meshes), and it's inconvenient. 

@patmarion I think we chatted about whether to have the `LOAD_ROBOT` command *only* delete any existing robot with the same robot ID number, rather than all robots. At the time, that seemed like a bad idea, because it leaves additional state in the visualizer. But I'm starting to wonder if it might be the right thing to do. It would certainly fix my immediate problem. 

But perhaps someone else has a better idea, or another use case we need to address? 

How about it? Is anyone else trying to visualize multiple robots simultaneously in the `drake-visualizer`?

liangfok, 2016-11-29:
I have in the past and will again soon. @naveenoid  has a PR that loads multiple models: 

https://github.com/RobotLocomotion/drake/pull/4107

We are treating all robots as if they are one since that is how they are treated in `RigidBodyTree`. I think supporting what you want will be very useful once we scale up the number of models in our simulations.

patmarion, 2016-11-30:
It's fine with me if you want to make the change so that LOAD_ROBOT only replaces the named robot.  You might want to check with more drake users than are watching this thread.  I dunno if the automotive team at TRI might load scenes with different robot names and be surprised that the old stuff isn't cleaned up.  To delete an old robot the user can select and press delete in the scene browser, or restart drake-visualizer.

jwnimmer-tri, 2016-11-30:
If we are changing the message semantics, I think it should be a new name.  Creating a few new protocol primitives to add & remove robots seems sensible.

/CC @david-german-tri

david-german-tri, 2016-11-30:
Perhaps I'm being obtuse, but does the load_robot message even include a robot ID right now?

https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_viewer_load_robot.lcm

liangfok, 2016-11-30:
Yes, it's hidden here:

https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_viewer_link_data.lcm#L5

patmarion, 2016-11-30:
By the way, what used to be called `robot_num` has been renamed to `model_instance_id` in RigidBodyTree.

rdeits, 2016-11-30:
What about sending the same message but on a new channel? For example, `DRAKE_VISUALIZER_ADD_ROBOT` instead of `..._LOAD_ROBOT`. Where `LOAD_ROBOT` keeps the current behavior (deleting all other robots), but `ADD_ROBOT` will only delete a robot that has the same ID number.

david-german-tri, 2016-11-30:
I don't think I agree with @jwnimmer-tri.  It doesn't make much sense that "load" and "reset" are munged together in `LOAD_ROBOT` right now, and the original proposal to separate them sounds good to me.  Changing message semantics are inevitable; we just need to make sure there's a migration path for both senders and receivers.  In this case that path is trivial, since as far as we know no senders are relying on the current behavior.  (If there are any, we can give them a reset message.)

All that said, @liangfok is the expert on this stuff at TRI, so I'm happy to defer to him.

rdeits, 2016-11-30:
@david-german-tri I agree that as far as I know, no one (except me) is currently using `drake-visualizer` with more than one robot ID number. That's because we've always used the trick that @liangfok mentioned in which we combined multiple robots into a singe rigid body tree/manipulator.

liangfok, 2016-11-30:
I like the idea of using separate `DRAKE_VISUALIZER_ADD_ROBOT` and `DRAKE_VISUALIZER_LOAD_ROBOT` channels. Consider also including a `DRAKE_VISUALIZER_REMOVE_ROBOT` channel as a counterpart to `DRAKE_VISUALIZER_LOAD_ROBOT`.

david-german-tri, 2016-11-30:
Seems like a weird use of the "channel" semantic to me, but then, I don't understand the point of semantic channels in LCM at all.  :shrug:

rdeits, 2016-11-30:
The benefit of using a channel instead of adding a flag to the message is that it does not break backwards compatibility for users of the old message type.

rdeits, 2016-12-01:
Ok, given that there's not a compelling reason to break the existing behavior, I will try the `ADD_ROBOT` approach.

PR #393 (2016-12-01): Support adding and removing robots in drakevisualizer
Fixes #389 

drakevisualizer now listens to `DRAKE_VIEWER_ADD_ROBOT` ~~and DRAKE_VIEWER_REMOVE_ROBOT~~. `ADD_ROBOT` has the same message type as the existing `LOAD_ROBOT`, but does not clear existing robots (unless an existing robot has the same robot ID number as the robot being added). ~~REMOVE_ROBOT just includes a single robot ID number to delete, or -1 to delete all robots.~~

~~Note: I am abusing the LCM `utime_t` here as the message type for `REMOVE_ROBOT`, in order to avoid creating a branch new LCM type just to send an integer. Thi [... cut]
closed 2016-12-01 by patmarion