RobotLocomotion/director #391: Copy affordance broken

wxmerkt, 2016-11-30:
```
copyAffordance
    aff = robotSystem.affordanceManager.newAffordanceFromDescription(desc)
AttributeError: 'FieldContainer' object has no attribute 'affordanceManager'
Traceback (most recent call last):
  File "/home/wxm/dev/oh-distro-private/software/build/lib/python2.7/dist-packages/director/robotviewbehaviors.py", line 490, in copyAffordance
    aff = robotSystem.affordanceManager.newAffordanceFromDescription(desc)
AttributeError: 'FieldContainer' object has no attribute 'affordanceManager'
Traceback (most recent call last):
  File "/home/wxm/dev/oh-distro-private/software/build/lib/python2.7/dist-packages/director/robotviewbehaviors.py", line 490, in copyAffordance
    aff = robotSystem.affordanceManager.newAffordanceFromDescription(desc)
AttributeError: 'FieldContainer' object has no attribute 'affordanceManager'
```

patmarion, 2016-11-30:
thanks for reporting.   I think this can be resolved by adding 'Affordances' to the list of declared dependencies of 'RobotViewBehaviors' in robotsystem.py

patmarion, 2016-11-30:
here https://github.com/RobotLocomotion/director/blob/master/src/python/director/robotsystem.py#L36

commit 5d53432f (2016-11-30): Merge pull request #392 from RobotLocomotion/wxm-fix-affordance-copy

Fix affordance-copy, closes #391
commit e336792a (2016-11-30): Fix affordance-copy, closes #391
closed 2016-11-30 by wxmerkt