RobotLocomotion/director #415: New drake visualizer interface

rdeits, 2017-01-02:
Continuing the discussion from https://github.com/rdeits/DrakeVisualizer.jl/issues/21#issuecomment-268728737

I'd like to propose a complete overhaul of the drake visualizer interface. The current drake-visualizer is great, but it has several issues that can't be avoided with its current architecture:

* clients must always send a LOAD message before sending a DRAW message. If the visualizer is ever restarted, the clients have no way of knowing, and will just keep sending useless DRAW messages. 
* the only organizational structure allowed is an integer robot_id_number and a string link name
* the only way to add new geometries or add new features to existing geometries is to pack floats into the float_data of the `viewer_geometry_data_t` lcmtype, which is extremely clunky
* drake-visualizer currently uses entirely different channels and code paths to draw
  * link geometries
  * lidar points
  * pointclouds
* that means that, for example, several lidars cannot be rigidly connected, and a pointcloud cannot be attached to a robot link

To solve all of these issues, I'm proposing several major changes:

* Create a response protocol for the visualizer, which at least indicates whether a given action (like DRAW or LOAD) was successful. 
  * More specifically, when a client tries to draw a link at a given pose, the visualizer may respond with a "no such link" message and the name of that link, so that the client can send an appropriate LOAD message
* Stop using LCM types for geometry data. The fact that new fields cannot be added to LCM types without breaking compatibility is a major annoyance. Instead, we can just pack a set of bytes that encode, e.g. a JSON string into an LCM message. This makes it trivial to add new options in the future, without breaking compatibility
* Use this increased type flexibility to make pointclouds and lidars just another geometry type. This makes it easy to add pointclouds which rigidly move together (by attaching them to the same link)
* Replace the "robot_id_num" and link name concepts with a general "path" concept. A path is a list of strings, and a link with path ["robot1", "group2", "link3"] is placed in the folder `robot1/group2/link3` in the visualizer. This reproduces all of the capability of the robot_id_num concept and also allows lots of other organizational schemes

I've done all of the above on my branch here: https://github.com/rdeits/director/tree/json-interface

Pointclouds, meshes, LIDARs, and geometric primitives are all working. The basic client-server protocol is working, so now the visualizer can be killed and restarted while a client is drawing, and everything will "just work" (i.e. the model will be re-loaded when necessary). It also makes most of the drake visualizer code quite a bit simpler, since we don't have separate code for pointclouds and lidars. 

It's possible to use the new interface without the client being smart: if you just blindly send LOAD and then DRAW (the way the clients do now), everything will work. You just won't have the benefit of being able to stop and start the viewer at will. 

But before I PR this, I want to make sure this is the right thing to do. I don't imagine we'll want to delete the current drake-visualizer. Instead, I think we should keep it (but deprecate it) and bring in the new interface as `drake-visualizer2` or similar. 

Also, I haven't noticed any performance issues from the switch from LCM types to JSON dicts. But if there is one, we can trivially swap out JSON for MsgPack with almost no code change (and we can support clients using both formats with no trouble).

dehann, 2017-01-02:
Hi @rdeits, I could maybe comment on JSON, LCM. I think there will ultimately be convergence on the issue serialization/de-serialization (we not there yet though). JSON, BSON and Protobufs (on REST layers) seem to be the big players. LCM, ROS, MOOS, MsgPack might be a bit smaller. I have seen people stack JSON strings into LCM types to get the benefits of logging, playback and spy. I've also seen Protobufs packed into LCM/ROS/MOOS. I think JSON is a good bet, and its pretty easy to record and play back a JSON string via LCM for demos etc.

BSON is the binary version of JSON. See [LibBSON.jl](https://github.com/pzion/LibBSON.jl).

I'm still getting up to speed with Director internals, but must complement you guys on the current Director+DrakeVisualizer experience.

From our previous discussion, and we can take this back to [rdeits/DrakeVisualizer.jl#21](https://github.com/rdeits/DrakeVisualizer.jl/issues/21#issuecomment-268728737), would drawing triads be better suited under geometries, or links or something else? Imagine wanting to draw a 1000 triads. I'm guessing an organization like ```poses/1```, ```poses/2...```?

Slaving pointclouds to a pose could conceptually be something like ```poses/1/pointcloud1```. Updating the transform on ```poses/1``` would automatically update everything lower down on that tree. Maybe there is also a geometry such as ```poses/1/leftcamfrustum``` which then also gets updated on the viewer side.

dehann, 2017-01-02:
Hardly related, but maybe a good reference. FreeCAD also uses VTK and Python. They have the idea of a [object tree](https://youtu.be/bosJdm2iEKw?t=291), and do [cool calculations and animations like this](https://www.youtube.com/watch?v=SMsN3pZITYE). FreeCAD is much more focused on the interaction between objects in the tree though, but the logic of the object tree with primitives is oddly useful. Basically supporting your suggestion:
> Replace the "robot_id_num" and link name concepts with a general "path" concept. A path is a list of strings, and a link with path ["robot1", "group2", "link3"] is placed in the folder [...]

rdeits, 2017-01-02:
@dehann OK, that's good to know. I'm glad JSON seems like a reasonable initial choice. 

I think that the right thing to do would be to make a triad another geometry primitive. The relevant JSON might look like:

```json
{
  "type": "triad",
  "lengths": [1, 1, 1]
}
```

We could also, optionally, add a "radius" argument to draw the triad as tubes with the given radius, with the default 0 radius corresponding to a triad made up of lines instead.

However, if you've got something like 1000 triads, that might overwhelm the object model. Director *can* easily draw that much geometry, but creating 1000 separate elements in the object model tree creates a lot of overhead. In that case, the right thing to do would probably be to have another geometry type like "triad_cloud":

```json
{
  "type": "triad_cloud",
  "translations": [[0, 0, 0], [1, 2, 3]],
  "rotations": [[1, 0, 0, 0], [1, 0, 0, 0]],
  "lengths": [1, 1, 1]
}
```

That way you can draw as many triads as you want without bogging down the director. The disadvantage would be that a triad could would be exactly one item in the object tree, so, for example, you could only toggle the visibility of the entire cloud. Is that OK?

And yes, agreed about the way slaving poses would work. In my mind, "poses/1" would be a "link" and "pointcloud1" and "leftcamfrustum" would be "geometries" attached to that link. Currently, I'm only supporting setting a transform for a leaf-path (that is, you could set the transform of "poses/1" but could not set the transform for "poses"). 

But the more I think about it, the more that leaf-path restriction feels totally arbitrary. We could do exactly what you're suggesting: let the user set the transform for *any* path, leaf or otherwise, and have it cascade down to all of the children of that path. That would let you do some cool things like:

* create `poses/1/pointcloud` and `poses/1/cam` 
* set the transform for `poses/1` to move both together
* set the transform for `poses/1/cam` to adjust the camera position calibration
* set the transform for `poses/1` to move both together while preserving their relative transforms
* set the transform for `poses` to move the entire collection of poses 

And I think it could actually make the code simpler to implement. Ok, I'll do that.

dehann, 2017-01-02:
@rdeits, what would be the best way to work with the transforms? If you could overload oplus, it might make sense to implement: transform for origin to ```poses``` as identity, then pose 1 would calculated by origin \oplus poses \oplus 1 to draw ```poses/1```. The transform for some geometry below pose 1 would be relative to ```pose/1``` and not the origin frame. Lets say ```poses/1/camfrustum``` geometry is then moved to position via 3 chained transforms: origin \oplus poses \oplus 1 \oplus camfrustum.

The advantage is a simpler iterator approach for all elements below the tree at that point. And the logic is that if any transform is updated, you simply iterate against all the /oplus paths below that point in the tree. Do you have code like that, or that I can with help with it somehow?

I like the idea triad primitive and addition of ~~~triad_cloud~~~triad_set(?). In Collections Render we usually draw to a set of poses, where visibility and parameters are adjust to the whole set. That feels about the same. I'll work on that.

rdeits, 2017-01-02:
@dehann I agree totally about the chained transforms. But I'm proposing having all of the actual chaining occur within Director. That is, you can just send a command like "draw poses/1 at transform T" and the chain of transforms below the tree at that point will all be updated automatically, so the drawn pose of `poses/1/camfrustum` will be the chained transform `T_poses + T + T_camfrustum`. We can mirror that structure on the Julia side by just borrowing @tkoolen's tree data structure: https://github.com/tkoolen/RigidBodyDynamics.jl/blob/master/src/tree.jl 

And yes, that's right--the triad cloud would be a leaf in the object tree.

sammy-tri, 2017-01-03:
I have a slight concern that switching to JSON could require pretty substantial amounts of bandwidth and parsing overhead for even modest pointclouds (640x480x32bpp and 30hz is a lot of text), with more concern if we've got multiple (or denser) sensors.  A quick survey (ok, reading some Wikipedia pages) indicates that we've got plenty of options for binary encoding of JSON, so perhaps that would satisfy and require no major code changes.

For robot poses I doubt we'd care about update rates which would matter with even a significant number of joints.

At any rate, my concern is not sufficiently well founded/researched to be considered an objection (just an observation).

rdeits, 2017-01-03:
That's a reasonable concern. I'll run some benchmarks and check how much better something like MsgPack or BSON would be.

jwnimmer-tri, 2017-01-03:
/CC @david-german-tri who has a very related project.

rdeits, 2017-01-03:
Ok, did some simple profiling, For a dataset consisting of 640*480 32-bit floats, JSON is predictably slow (hundreds of milliseconds in Python). MsgPack with the numpy extensions is stupidly, ridiculously fast (hundreds of *microseconds* to encode or decode).

rdeits, 2017-01-03:
Note, however, that I think msgpack-numpy does this by cheating: it serializes arrays as raw byte strings, and therefore does not actually use the msgpack array format (or the msgpack format for each number in the array).

rdeits, 2017-01-03:
@dehann I've got the arbitrary-chains-of-transforms version working. I haven't updated `DrakeVisualizer.jl` yet, but if you'd like to play around with the Python version, I think you can get a good idea of what's now possible. 

If you're interested, you should check out Director on the [rdeits/path-transforms](https://github.com/rdeits/director/tree/path-transforms) branch. All you need to do is build director, launch the visualizer with `build/install/bin/drake-visualizer`. To try drawing things, you can use the demo python script I put together. Just open a new terminal, cd to `director` and then run `python newviewer.py`. That script will eventually get cleaned up as a proper unit test, but for now you can hack it however you want. Currently, `newviewer.py` demonstrates loading several different geometries and then setting transforms for various paths. 

Let me know if you have any issues. This is all brand-new, so there are probably bugs or things that need clearer names.

rdeits, 2017-01-04:
@dehann and now I've got the Julia bindings working well enough for a basic demo. There's a new [path-transforms](https://github.com/rdeits/DrakeVisualizer.jl/tree/path-transforms) branch with a [demo notebook](https://github.com/rdeits/DrakeVisualizer.jl/blob/path-transforms/viewer2.ipynb) that should let you play around with the functionality. 

Again, all of this is still subject to cleanup and improvements, but the basic capabilities should be working now.

david-german-tri, 2017-01-07:
> /CC @david-german-tri who has a very related project.

Thanks!  I'm a little late but will put in a couple of cents anyway.  The "related project" jwnimmer-tri mentions is an effort to standardize a rendering interface within TRI.  The basic principle is to draw a sharp line between "dynamics engines" like Drake and "rendering engines" like Visualizer.  The only information that crosses that line is poses for named models.  It's up to each engine to associate geometry and other physical properties with the names.

So, if you're open to a PR that adds a command-line switch to look for geometry on disk instead of in the message, I'll send one eventually.  If not, we can always handle that in an adapter layer.  Beyond that, I think this interface provides all the features I need, as well as many that I don't, so that's cool.

Have a good weekend!

rdeits, 2017-01-07:
Thanks @david-german-tri, that sounds good. The new viewer actually already does support loading meshes from file: https://github.com/RobotLocomotion/director/pull/420/files#diff-c70a0cd7c4e665d31946b5a54fdc9e59R102 (I just haven't added a test or demo for that feature yet, sorry). Does that address your use case?

PR #420 (2017-01-20): Shiny new drake visualizer
Copied from #419 so @patmarion can also push to the branch. 

@patmarion this will probably need some work before it's ready to merge, but I wanted to get started on the PR now. 

This implements the interface described in #415 (closes #415)

I've split the old `drakevisualizer.py` into `drakevisualizer.py` and `drakevisualizerapp.py`. I've also created the new interface in `drakevisualizer2.py`. Additionally, I've created `drakevisualizerapp1.py` and `drakevisualizerapp2.py` which import the relevant modules to construct the v1 and v2 apps. And I've added a command  [... cut]
closed 2017-01-20 by patmarion