RobotLocomotion/director #446: Director cannot handle collections messages at very high rate

mauricefallon, 2017-02-03:
Not going to fix it now, reporting anyway:

Director locks up when receiving collection messages (from pronto in debug mode) at more than 300Hz or so. Need to disable collections in startup

patmarion, 2017-02-03:
It's likely because of these lines:

https://github.com/RobotLocomotion/director/blob/master/src/python/director/lcmcollections.py#L207

The way collections was implemented in Director (by following lcmgl.py's implementation) put the lcm subscribers in Python.  But as we know, python cannot decode messages at a fast rate.  The best fix would be to move the lcm subscribers to threaded c++.  If you feel strongly about it I can help.

If you removed the lines calling `setNotifyAllMessagesEnabled(True)` then it might resolve the lock up issue, but it means some messages will be skipped.

mauricefallon, 2017-02-04:
This is used for analysis - so allowing messages to be skipped is not a suitable option.
I think the correct option is to move decoding to c++. I'll come back to this when I have the cycles.