RobotLocomotion/director #464: Request: versioned binary names

rdeits, 2017-02-15:
@patmarion how would you feel about either putting the git SHA in the director binary names and keeping at least some prior versions? It's difficult for DrakeVisualizer.jl to reliably use the binaries since I can't reliably download the appropriate binary version for a given version of my package. 

If keeping all the git sha's is too much clutter, maybe we could at least bump the version numbers and keep old versions around?

patmarion, 2017-02-15:
sure im happy to include the sha1.  The package name is determined here:

https://github.com/RobotLocomotion/director/blob/master/distro/package/make_linux_package.sh#L12

just need to also query `ver.commitHash()`.  I can work on it later, but if you want to do it yourself please go for it!  Also update make_mac_package.sh.

Packages are uploaded to a staging area everytime travis-ci builds director/master.  Then I manually scp packages from the staging area to the public www only on demand, usually when you add a new feature you want.  I don't mind having them pile up, ill eventually automate it.

rdeits, 2017-02-15:
Ok, I'll work on that now.

PR #465 (2017-02-15): add commit hash to package names
resolves #464
closed 2017-02-15 by patmarion