RobotLocomotion/director #470: Should precompiled binaries require numpy?

rdeits, 2017-02-19:
Or should it be included with the precompiled package? 

ref. https://github.com/rdeits/DrakeVisualizer.jl/issues/35

patmarion, 2017-02-19:
Current setup:

  - mac packages: all dependencies are bundled, users don't need to `brew install` anything, nor does the package use apple's system python or numpy.

  - linux packages: only bundle what is compiled, users are expected `apt-get install` everything else

I did it this way since it is very easy for users to `apt-get install` the dependencies with a single command, but it's not easy for mac users to fumble with homebrew.  The set of required `apt-get` packages probably isn't clearly documented... we could fix that, and provide a `install_dependencies.sh` script in the package to make it easy.

rdeits, 2017-02-19:
Ok, that all makes sense. I'm fine with having users apt-get install things manually, I just didn't know what the list of requirements was.

closed 2017-02-19 by rdeits

patmarion, 2017-02-19:
Note, this is the list of development dependencies on ubuntu (copied from the top-level README):

`sudo apt-get install cmake libvtk5-qt4-dev libeigen3-dev python-dev python-vtk python-numpy`

Installing those is sufficient to run the director binaries.  But I haven't documented the runtime dependencies.  You could get away with installing a more restricted set.