RobotLocomotion/director #494: Support point size when drawing point clouds with remote tree viewer

gizatt, 2017-03-29:
I'm not 100% sure this isn't supported by the channel feature, but I don't think it is. 

Not high priority, just writing this down because it would be useful.

patmarion, 2017-03-29:
+1.  Also, line width would be a nice to add.

On the director side, `Point size` is an exposed display property.  Line width is not an existing property but it is easily added and I'd be happy to help.

dehann, 2017-03-30:
Hi, maybe I can mention something related. I'm drawing point clouds which end up at the leaves of remote tree viewer. However, to change the size of points, you'd have to update them all individually down the tree one by one... Is there maybe a shortcut for this or a global modification field?

rdeits, 2017-03-30:
Sounds like we might need a new command like `setproperties` with a path and a dictionary of properties, which would then recursively set those properties for all geometries below the path.

patmarion, 2017-05-15:
fyi, i recently merged a branch which adds a `Line Width` property in director.  So if we wanted this could also a display property in remote tree viewer.  This lets you draw thicker lines without having to draw lines as cylinders.

patmarion, 2017-05-16:
you can also set `Line Width` on the triad objects, which is convenient for making them more visible.