RobotLocomotion/director #503: Cannot create GLX Context

gizatt, 2017-04-20:
I'm exploring the many instances of this problem reported online (it originates in VTK), but for documentation's sake:

After no modifications I'm aware of (no reboot, no apt-get upgrade, no rebuild), `drake-visualizer` and `directorPython testMainWindowApp.py` (and probably others) have stopped launching with error:

```
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x263af40): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x263af40): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 631
vtkXOpenGLRenderWindow (0x263af40): Cannot create GLX context.  Aborting.

Aborted (core dumped)
```

This has persisted after a clean rebuild (of spartan) and a reboot.

While I search and try stuff, any ideas?

patmarion, 2017-04-20:
Can you run glxgears?  Can you post output of glxinfo?

I have only seen this after modifying the video driver, and it resolves after a reboot.

gizatt, 2017-04-20:
glxgears works without warnings, glxinfo is at https://pastebin.com/izg6L2BR

Just reinstalled my video driver and rebooted and the problem is persisting; then purged out vtk 5.8, reinstalled, and rebuilt director (make -B) and it's persisting

patmarion, 2017-04-21:
How is it picking up vtk-5.8 from this path: `build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx`?  Is that what you expect?  Do you also have vtk system installed?

gizatt, 2017-04-21:
I'm pretty confused by that path -- a "buildd" folder doesn't show up
anywhere in my system.

I'm not sure what you mean by vtk system?

gizatt, 2017-04-21:
I can't reproduce this in standalone drake-building-director; only in Spartan. I'm guessing it could be that the Spartan Director build is pulling in a conflicting version of a library...

manuelli, 2017-06-12:
I fixed this by setting the USE_SYSTEM_VTK=OFF option.

patmarion, 2017-06-12:
There are two things to double check:

1) which version you compiled and linked with
2) which version of vtk gets loaded at runtime when you `import vtk`

They need to be the same.

QinZiwen, 2018-03-11:
@gizatt  Is this problem solved?

gizatt, 2018-03-11:
IIRC this was a VTK version mismatch (system vs director-or-drake-source-installed) that I haven't had trouble with since

QinZiwen, 2018-03-12:
@gizatt   Sorry, I didn't understand what you meant

QinZiwen, 2018-03-12:
Vtk no special requirements, like:
```
In general VTK tries to be as portable as possible; the specific configurations below are known to work and tested.

VTK supports the following C++11 compilers:

Microsoft Visual Studio 2013 or newer
gcc 4.8.3 or newer
Clang 3.3 or newer
Apple Clang 5.0 (from Xcode 5.0) or newer
Intel 14.0 or newer

VTK supports the following operating systems:

Windows Vista or newer
Mac OS X 10.7 or newer
Linux (ex: Ubuntu 12.04 or newer, Debian 4 or newer)
```

gizatt, 2018-03-12:
Right, but Director expects a certain vtk version (6 or 7, I think? see the
director homepage) while your system might have another version installed.
If you run director and it loads vtk dynamic libraries that are a different
version than it director was built with, this problem might come up.