RobotLocomotion/director #505: Test failure with VTK master and Qt 5.7.1

sankhesh, 2017-04-24:
```python
test 13
    Start 13: test_testTransformations

13: Test command: <build_dir>/install/bin/directorPython "<source_dir>/src/python/tests/testTransformations.py" "--testing" "--data-dir" "<source_dir>/../../../drc-testing-data" "--output-dir" "<build_dir>/src/director-build/Testing/Temporary"
13: Environment variables:
13:  COVERAGE_PROCESS_START=<build_dir>/src/director-build/python-coverage/coverage.conf
13: Test timeout computed to be: 1500
13: <build_dir>/install/lib/python2.7/dist-packages/director/thirdparty/transformations.py:1901: UserWarning: failed to import module _transformations
13:   warnings.warn("failed to import module %s" % name)
13: [[-0.99185063 -0.12094519  0.04005741  0.34950866]
13:  [ 0.09957138 -0.53969891  0.83594894  0.51591451]
13:  [-0.07948507  0.83312505  0.54734338  0.44465101]
13:  [ 0.          0.          0.          1.        ]]
13: [[-0.99185063 -0.12094519  0.04005741  0.34950866]
13:  [ 0.09957138 -0.53969891  0.83594894  0.51591451]
13:  [-0.07948507  0.83312505  0.54734338  0.44465101]
13:  [ 0.          0.          0.          1.        ]]
13: (2.487323779613272, 0.13340592974659932, 0.4773399184643872)
13: (1.299095545242626, 0.01942238985095687, 2.161023278876459)
13: Traceback (most recent call last):
13:   File "<source_dir>/src/python/tests/testTransformations.py", line 129, in <module>
13:     testEuler()
13:   File "<source_dir>/src/python/tests/testTransformations.py", line 85, in testEuler
13:     assert np.allclose(rpy, rpy2)
13: AssertionError
1/1 Test #13: test_testTransformations .........***Failed    2.57 sec
```

patmarion, 2017-04-25:
> 13: <build_dir>/install/lib/python2.7/dist-packages/director/thirdparty/transformations.py:1901: UserWarning: failed to import module _transformations

this might be the problem.  when you configure director, do you see this warning message:

`Optimized transformations C extension module is disabled because numpy is not found.`

sankhesh, 2017-04-25:
> do you see this warning message:
> 
> Optimized transformations C extension module is disabled because numpy is not found.

I do. However, its strange that the test passes when director is compiled against Qt 5.6.1.

sankhesh, 2017-04-26:
This issue was resolved once I installed numpy in the right place. I had numpy installed in a conda environment which was in my PYTHONPATH and that might getting picked up with other tests (I guess).

closed 2017-04-26 by sankhesh

patmarion, 2017-04-26:
yeah, unfortunately there seems to be a bug in transformations.py.  Normally it loads a c module which works correctly, but the python code that runs when the c module is not available seems to have a bug.  That's why I wrote that test!