RobotLocomotion/director #524: Does not work with VTK 7 and PCL enabled due to build errors in PointCloudLibraryPlugin

gizatt, 2017-06-16:
To be specific, build errors arise in `PointCloudLibraryPlugin`, e.g.
```
threshold->SetInput(inputCopy);
```
(which I know changed from VTK 5 to VTK 6+).

Just FYI!

patmarion, 2017-06-16:
I have a branch where i am working on updating the full Director superbuild including PCL to work with upgraded VTK.  In the meantime, non of this is working and you should use the `director/master-vtk5` branch.  See #519 Thanks!

gizatt, 2017-06-16:
Great, glad it's on the radar! Excited for some VTK 7 features soon.

patmarion, 2017-06-16:
Yeah, so I've actually had it fully working locally, no issues.  Just a matter of getting things cleaned up and committed, which will happen along with things like: drake cmake build updating to director/master and spartan updating to drake/master, etc.

patmarion, 2017-08-13:
Fixed here: https://github.com/patmarion/PointCloudLibraryPlugin/pull/1

closed 2017-08-13 by patmarion