RobotLocomotion/director #526: Segmentation fault when trying to `dlclose()` libddApp.dylib on OSX (from Julia)

rdeits, 2017-06-22:
Since this only happens from Julia, I'm not sure that this is really your problem, but this used to work and seems to have broken sometime in the transition from vtk5 to vtk7. To reproduce, download the pre-build binaries of version `0.1.0-130-g4109097` (the latest) and cd to the `lib` folder, then run:

```
julia -e 'h=Libdl.dlopen_e("libddApp.dylib", Libdl.RTLD_LAZY); Libdl.dlclose(h)'
```

on my machine (macOS 10.11), I get:

```
signal (11): Segmentation fault: 11
while loading no file, in expression starting on line 0
_ZN13vtkCollection13DeleteElementEP20vtkCollectionElement at /Users/rdeits/.julia/v0.6/DrakeVisualizer/deps/usr/Director.app/Contents/MacOS/lib/libvtkCommonCore-7.1.1.dylib (unknown line)
_ZN13vtkCollection14RemoveAllItemsEv at /Users/rdeits/.julia/v0.6/DrakeVisualizer/deps/usr/Director.app/Contents/MacOS/lib/libvtkCommonCore-7.1.1.dylib (unknown line)
_ZN13vtkCollectionD2Ev at /Users/rdeits/.julia/v0.6/DrakeVisualizer/deps/usr/Director.app/Contents/MacOS/lib/libvtkCommonCore-7.1.1.dylib (unknown line)
_ZN26vtkObjectFactoryCollectionD0Ev at /Users/rdeits/.julia/v0.6/DrakeVisualizer/deps/usr/Director.app/Contents/MacOS/lib/libvtkCommonCore-7.1.1.dylib (unknown line)
_ZN16vtkObjectFactory22UnRegisterAllFactoriesEv at /Users/rdeits/.julia/v0.6/DrakeVisualizer/deps/usr/Director.app/Contents/MacOS/lib/libvtkCommonCore-7.1.1.dylib (unknown line)
__cxa_finalize_ranges at /usr/lib/system/libsystem_c.dylib (unknown line)
exit at /usr/lib/system/libsystem_c.dylib (unknown line)
Allocations: 831262 (Pool: 830227; Big: 1035); GC: 0
[1]    19797 segmentation fault  julia -e
```

The reason this matters is that Julia uses `dlopen()` and `dlclose()` to check if a library is installed, which is triggering this segfault. My workaround is instead to tell Julia to look at the `libvtkDRCFilters` library, which doesn't segfault.

patmarion, 2017-06-22:
Thanks for testing the new binaries!

I will try to reproduce on my mac and see if I can figure it out.

rdeits, 2017-06-22:
Thanks for making them! I'm updating DrakeVisualizer.jl to use the newest version of Director on all platforms right now: https://github.com/rdeits/DrakeVisualizer.jl/pull/55

patmarion, 2017-06-22:
Sounds good.  Would you be able to update to Python 3 as well?  Any reason you need to continue using Python 2?  I have a Python 3 branch nearly ready to go, I just need to generate mac vtk-python3 packages so that it can pass tests on Travis-CI.

rdeits, 2017-06-22:
Yeah, there's nothing blocking that switch (other than having binaries released). All my communication is over the LCM channel, so I probably won't even notice.

patmarion, 2017-06-23:
I was able to reproduce the crash.  I will let you know if I get a fix for it.  In the meantime, I think you have a viable workaround so I am not treating this as a high priority.

rdeits, 2017-06-23:
Yeah, my workaround is fine. Thanks for taking a look 🙂