RobotLocomotion/director #538: Background Light toggle

gtinchev, 2017-07-27:
Hi,

I'm adding some custom functionality as part of a customized panel and I would like to toggle the background light in certain scenarios (via the code).

I am able to get it to dark, but not to light again thereafter. I could see that the properties of "view options" - "background color" and "background color 2" change their values. However, I could not see any changes made to the grid.

The code I'm using:

```
grid = om.getOrCreateContainer('grid')
viewOptions = vis.ViewOptionsItem(self.view)
viewBackgroundLightHandler = self.viewcolors.ViewBackgroundLightHandler(viewOptions, grid, app.getToolsMenuActions()['ActionToggleBackgroundLight'])
viewBackgroundLightHandler.action.trigger()
```

Could you please assist?

patmarion, 2017-07-27:
If there a reason you decided to construct the application objects yourself rather than use the function `mainwindowapp.construct()` to construct everything?  Are you running your code with:  `directorPython script.py` or `drake-visualizer --script scripy.py` because these are different commands.

gtinchev, 2017-07-27:
The only reason for not using mainwindowapp is that I do not have access to viewBackgroundLightHandler from the panel. I believe it lives in viewcolors.

Therefore, I cannot use the following code either:
```
# black background
viewBackgroundLightHandler.setEnabled(True)

# user defined background (light gray default)
viewBackgroundLightHandler.setEnabled(False)
```

However, in Director, if I use the interactive python interface (F8) and type:

```
viewBackgroundLightHandler.action.trigger()
```
It does the job. So all I'm looking after is a way to access this object from my code.

Thanks!

patmarion, 2017-07-27:
Perhaps you can pass the existing instance to the constructor of the panel?  The instance is probably avaialable to the code that is constructing the panel.  Can you post what that code looks like?

gtinchev, 2017-07-27:
startup.py:

```
import diftdriver
import diftdriverpanel

from director import tasklaunchpanel
from director import applogic
from director import teleoppanel
import director.objectmodel as om

from director import viewcolors
from director import cameraview

def startup(robotSystem, globalsDict=None):
    rs = robotSystem

    diftDriver = diftdriver.DiftDriver()
    diftDriverPanel = diftdriverpanel.init(diftDriver, viewcolors, applogic.getDRCView())

    # Export variables to globals so that they can be accessed from the console
    if globalsDict is not None:
        globalsDict['dift'] = diftDriver
        globalsDict['diftDriverPanel'] = diftDriverPanel
```

diftdriverpanel.py:

```
import PythonQt
from PythonQt import QtCore, QtGui, QtUiTools
from director import applogic as app
from director.utime import getUtime
from director import transformUtils
from director import lcmUtils
from director import objectmodel as om
from director import visualization as vis
from director import pointpicker
from director import cameracontrol

import json
import os
import vtk
import glob
import bot_core as lcmbotcore
import drc as lcmdrc

def addWidgetsToDict(widgets, d):

    for widget in widgets:
        if widget.objectName:
            d[str(widget.objectName)] = widget
        addWidgetsToDict(widget.children(), d)

class WidgetDict(object):

    def __init__(self, widgets):
        addWidgetsToDict(widgets, self.__dict__)

class DiftDriverPanel(object):

    def __init__(self, driver,viewcolors, view):
        self.driver = driver
        self.view = view

        loader = QtUiTools.QUiLoader()
        uifile = QtCore.QFile(os.path.join(os.path.dirname(__file__), 'ui/ddDiftDriverPanel.ui'))
        assert uifile.open(uifile.ReadOnly)

        self.widget = loader.load(uifile)
        self.widget.setWindowTitle('Dift Panel')
        self.ui = WidgetDict(self.widget.children())

        # ui buttons hereafter
        self.ui.overheadButton.connect('pressed()', self.overheadButton)
        # ...
       
        self.viewcolors = viewcolors

    # irrelevant functions...

    def triggerBackgroundLight(self):
        grid = om.getOrCreateContainer('grid')
        viewOptions = vis.ViewOptionsItem(self.view)
        viewBackgroundLightHandler = self.viewcolors.ViewBackgroundLightHandler(viewOptions, grid, app.getToolsMenuActions()['ActionToggleBackgroundLight'])

    # more irrelevant functions

def _getAction():

    actionName = 'ActionDiftDriverPanel'
    action = app.getToolBarActions().get(actionName)

    if action is None:

        icon = QtGui.QIcon(os.path.join(os.path.dirname(__file__), 'images/map_icon.png'))
        assert os.path.isfile(os.path.join(os.path.dirname(__file__), 'images/map_icon.png'))

        action = QtGui.QAction(icon, 'Dift Panel', None)
        action.objectName = 'ActionDiftDriverPanel'
        action.checkable = True

        mainWindow = app.getMainWindow()
        toolbar = mainWindow.panelToolBar()

        toolbar.insertAction(toolbar.actions()[0], action)

    return action

def init(driver, viewcolors, view):

    global panel
    global dock

    panel = DiftDriverPanel(driver, viewcolors, view)
    dock = app.addWidgetToDock(panel.widget, action=_getAction())
    dock.hide()

    return panel

```

diftdriver.py:

```
import copy
import os
import math
import glob
import vtk

from director import ioUtils
from director import lcmUtils
from director.utime import getUtime
import director.objectmodel as om
from director import visualization as vis
from director import transformUtils
from director import segmentation

from debugVis import DebugData

import matplotlib.pyplot as plt
import numpy as np
import drc as lcmdrc
import bot_core as lcmbotcore

class DiftDriver(object):

    def __init__(self):
        #custom logic

def init():

    global driver
    driver = DiftDriver()

    return driver

```

patmarion, 2017-07-27:
I asume that the function in your `startup.py` is called like `startup(robotSystem, globals())`.  So that means that inside the function you can access the `viewBackgroundLightHandler` as:

```
viewBackgroundLightHandler = globalsDict['viewBackgroundLightHandler']
```

And then pass that variable in to the constructor of your panel.

gtinchev, 2017-07-27:
Worked like a charm. Did not realize vewBackgroundLightHandler is a global variable. Thanks!

closed 2017-07-27 by gtinchev

patmarion, 2017-07-27:
Great!  Yes, anything variable you can access in the Python console in Director should also be available in the `globalsDict` of that script.