RobotLocomotion/director #540: "No rule to make target" error when building

GianMassimiani, 2017-07-30:
I am completely new to Linux. I have installed a fresh Ubuntu 16.04 version and I have tried to build the Director. However I am getting this error:
```
make[6]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libSM.so', needed by 'lib/libvtkDRCFilters.so'.  Stop.
make[6]: Leaving directory '/home/gian/git/director/build/src/director-build'
CMakeFiles/Makefile2:1123: recipe for target 'src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all' failed
make[5]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
make[5]: Leaving directory '/home/gian/git/director/build/src/director-build'
Makefile:138: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/gian/git/director/build/src/director-build'
CMakeFiles/director.dir/build.make:116: recipe for target 'src/director-stamp/director-build' failed
make[3]: *** [src/director-stamp/director-build] Error 2
make[3]: Leaving directory '/home/gian/git/director/build'
CMakeFiles/Makefile2:71: recipe for target 'CMakeFiles/director.dir/all' failed
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[2]: Leaving directory '/home/gian/git/director/build'
Makefile:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/gian/git/director/build'
Makefile:22: recipe for target 'superbuild' failed
make: *** [superbuild] Error 2
```
I noticed there is no `libSM.so` in `/usr/lib/x86_64-linux-gnu/`, but I found a `libSM.so.6` and `libSM.so.6.0.1`. Any idea on how to solve this? Thanks

patmarion, 2017-07-30:
Thanks for posting.  I recently updated the list of required dependencies and I think I am missing some items.  Please try installing this set:

`sudo apt-get install build-essential cmake libqt4-dev python-dev python-numpy libgl-dev libglu-dev libxext-dev libxt-dev`

If you'd like, you can download precompiled binaries of Director so that you don't have to compile from source:

https://bintray.com/patmarion/director/director/0.1.0-153-ge0a5a7b#files

GianMassimiani, 2017-08-01:
@patmarion thanks for replying. I managed to solve in this way:

`sudo rm -f /usr/lib/x86_64-linux-gnu/libSM.so`
`sudo ln -s /usr/lib/x86_64-linux-gnu/libSM.so.6 /usr/lib/x86_64-linux-gnu/libSM.so`

However, I am very new to Linux so I have one more question. After building do I have to install it? Because if I type `make install` inside `director/build` nothing happens.

patmarion, 2017-08-01:
Ok, glad you got it working.  But I would not recommend the solution you posted here, you should never `sudo rm` anything from `/usr/lib` without a very good reason.  I suspect that you were just missing the development version of the library which may have been installed by one of the dependencies I listed above.

closed 2017-08-01 by patmarion