RobotLocomotion/director #541: vtkOBJReader fails at reading .obj files

GianMassimiani, 2017-08-01:
I am trying to read some .obj files using `ioUtils.readPolyData`. For example this one:

`polyData = ioUtils.readPolyData('Porsche_911_GT2.obj')`

However I keep getting errors like this:
```
ERROR: In /root/VTK-7.1.1/IO/Geometry/vtkOBJReader.cxx, line 271
vtkOBJReader (0x2243970): Error reading 'v' at line 3
```
I have tried with different .obj files, but they all raise errors of that type. I also tried to open them with MeshLab and re-save them as .obj, but it didn't work. I went searching for that `/root/VTK-7.1.1/IO/Geometry/vtkOBJReader.cxx` file but I soon realized that there is no `VTK-7.1.1` directory inside `/root`. I am wondering if I installed `the Director` correctly, as I did it in two different ways: 
- `git clone https://github.com/RobotLocomotion/director.git` and then used `make superbuild`
- download from https://bintray.com/patmarion/director/director/0.1.0-153-ge0a5a7b#read and extract into a folder on desktop.

It might seem a stupid thing (sorry it's the first time I use Linux), but in both cases I didn't `install` anything (in first case I only used `make superbuild`, in second case only extracted the .tar.gz on desktop). I am wondering if `director` and/or `VTK` have been installed at all. Any idea on this? Thanks

patmarion, 2017-08-01:
Can you post your file?  I might have seen this before, and it and might have to do with locale.  Do you use `.` or `,` as separators in floating point numbers?

Your install is correct.  The error message tells you the VTK source file where the error originates, but that doesn't mean you have the source file on your computer, since you downloaded binaries only.

GianMassimiani, 2017-08-01:
@patmarion You are right, it has to do with that. For example these are the first few lines of the .obj file:

```
# Wavefront OBJ exported by MilkShape 3D

v -0,830351 -0,359413 -0,750083
v -0.825420 -0.361854 -0.690273
v -0.833710 -0.363075 -0.752827
v -0.821758 -0.359107 -0.690581
v -0.832860 -0.353614 -0.068623
```
If I change `.` with `,` then `vtkOBJReader` reads the file. Does it have to do with the fact that I am in Italy and we use "," instead of "." for numbers? What should I do at this point, edit the .obj file or what else?

jhidding, 2018-09-07:
This issue is still not resolved with Vtk version 8 (running from Mayavi).

patmarion, 2018-09-07:
Hi, thanks for the update.  It sounds to me like an issue that would be best fixed in VTK, so you may want to post to that bug tracker too.

janklimaszewski, 2020-03-17:
Hi, I resolved this error on my Ubuntu 18.04 system by changing the locale settings. 
1. Run in terminal: 
`sudo update-locale LC_NUMERIC=C`
2. Log out and then log in.