RobotLocomotion/director #543: Alpha issue: inverted geometry

tkoolen, 2017-08-07:
When alpha is one, all is good:
[image]

But if I set alpha to 0.99, the geometry becomes inverted:
[image]

As a side note, there's an additional alpha under the 'Color' part of the properties panel that ranges from 0 to 255. It seems to do nothing.

I believe @siyuanfeng-tri and @TristanThrush were experiencing similar inversion issues, but possibly unrelated to alpha.

patmarion, 2017-08-07:
The geometry does not look natural with alpha=0.99 because of the drawing order of triangles.  To correctly visualize transparent geometry Director needs to enable VTK's depth peeling rendering feature.  It is currently disabled for 1) support of antialiasing on more systems 2) performance.  It would be nice to add a toggle to enable depth peeling, currently it can only be done by modifying C++ code and recompiling.

If you set alpha to something much lower like 0.10 the effect will not be very bad.  alpha=0.99 is the worst case scenario.

Sorry for the confusing about the 4-component color property editor.  Only the 3 RGB components are actually used.

patmarion, 2017-08-07:
I think there is also a bug where the alpha value can get set to 0.999 but the spinbox editor rounds it up and displays 1.0.  If you manually enter 1.0 then it takes the correct value.

tkoolen, 2017-08-07:
Thanks for the quick response. 

> I think there is also a bug where the alpha value can get set to 0.999 but the spinbox editor rounds it up and displays 1.0. If you manually enter 1.0 then it takes the correct value.

That could explain why Siyuan and Tristan were having these issues.