RobotLocomotion/director #561: lcmUtils.addSubscriber doesn't work without specifying messageClass argument

manuelli, 2017-09-26:
The following snippet only triggers the `onEstRobotState` callback, not the `onEstRobotStateOld` callback when LCM traffic on `EST_ROBOT_STATE` is being published.

```
from director import consoleapp
from director import lcmUtils
from director import robotstate
from director import drcargs
import drake as lcmdrake
import bot_core as lcmbotcore
import numpy as np
import argparse

def onEstRobotState(msg):
    print "got an EST_ROBOT_STATE msg"

def onEstRobotStateOld(msg):
    print "got an EST_ROBOT_STATE ms using old subscriber signature"

if __name__ == "__main__":

    # works
    subscriber = lcmUtils.addSubscriber(
    'EST_ROBOT_STATE', lcmbotcore.robot_state_t, onEstRobotState)
    consoleapp.ConsoleApp.start()

    # DOESN'T WORK
    # this is the old signature
    subscriber = lcmUtils.addSubscriber(
    'EST_ROBOT_STATE', callback=onEstRobotStateOld)
    consoleapp.ConsoleApp.start()

```

patmarion, 2017-09-27:
if you don't specify `messageClass` then your callback must have the signature `func(bytes,  channel)` instead of `func(msg)`.

closed 2017-09-27 by patmarion

manuelli, 2017-09-27:
Ok, it fails silently however if the callback doesn't have the `func(bytes,channel)` signature however. Something to be aware of.

patmarion, 2017-09-27:
yeah sorry :(  It's a PythonQt issue that I've been aware of for years and never bothered to fix.