RobotLocomotion/director #576: dealing with vtk6 from kinetic - director initial setup & installation error

ahundt, 2017-12-15:
The documentation says it will install vtk for me if it isn't there:
```
On Ubuntu the build does not require VTK to be installed. A compatible version of VTK will be downloaded (precompiled binaries) at build time.
```

However I'm getting the following error:
```

-- Found Qt4: /usr/bin/qmake (found version "4.8.7")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.12", m
inimum required is "2.7")
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.12", minimum required is "2
.7")
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found Eigen: /usr/include/eigen3
-- Eigen found (include: /usr/include/eigen3)
CMake Error at /usr/local/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message
):
  Could NOT find PythonQt (missing: PYTHONQT_INCLUDE_DIR PYTHONQT_LIBRARY)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSA
GE)
  cmake/modules/FindPythonQt.cmake:23 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/app/CMakeLists.txt:1 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ahundt/src/RobotLocomotion/director/build/CMakeFiles/CMakeOutput.log".
See also "/home/ahundt/src/RobotLocomotion/director/build/CMakeFiles/CMakeError.log".
```

It seems I have the vtk6 versions that come with ros kinetic:

```

± dpkg-query -L vtk-dev
-> [1]
ahundt@femur|~/src/robotics_setup on master!?
± apt list --installed | grep vtk
libvtk-java/xenial,now 5.10.1+dfsg-2.1build1 amd64 [installed,automatic]
libvtk5.10/xenial,now 5.10.1+dfsg-2.1build1 amd64 [installed,automatic]
libvtk6-dev/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6-java/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6-qt-dev/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6.2/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
libvtk6.2-qt/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
python-vtk6/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
tcl-vtk6/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
vtk6/xenial,now 6.2.0+dfsg1-10build1+debian11.1+osrf1 amd64 [installed,automatic]
```

As kinetic is a LTS release I'd prefer not to disrupt my current ROS installations. 

Could you advise me on the workaround you recommend?

Thanks!

patmarion, 2017-12-16:
Pre-compiled director binaries can be downloaded here:

https://bintray.com/patmarion/director/director/0.1.0-234-g74cea84#files

If you want to compile from source, that should also work and it will download the correct version of vtk.  I think you may have configured using the wrong CMakeLists.txt.  You should use the superbuild CMakeLists.txt, see instructions at: 

https://github.com/robotlocomotion/director#building

Here is where the option to use precompiled vtk vs. system vtk is defined:

https://github.com/RobotLocomotion/director/blob/master/distro/superbuild/cmake/externals.cmake#L19

mauricefallon, 2018-01-11:
Hi @ahundt were you able to resolve this issue?

I'm trying to install director using the pcl version (1.7) provided by ROS which is compiled against vtk6.2

patmarion, 2018-01-11:
I'm not actually sure that Director works with vtk 6.2.  I know it works with VTK 5.8 (the last 5.x release) and works on vtk6.3 thru vtk8.  I think ROS kinectic's vtk6.2 is not a great release version.  But you should be able to build director against it's own internal version of VTK.

@mauricefallon are you building director standalone or as part of an openhumanoids build?

mauricefallon, 2018-01-12:
I have all director's dependencies building in catkin including Drake. The code still uses LCM but its now ROS compatible. I want to make director fully catkin integrated and perhaps ROS integrated in future.

Its only a few CMake commands here and there. That works just fine with perception disabled. This uses your precompiled vtk7.

1) The problem with USE_PERCEPTION is basically PCL. Kinetic provides pcl1.7 compiled against vtk6.2. vtk6.2 uses qt5.
qt4 and qt5, vtk5 and vtk6 are are system installed with Kinetic. 
ddApp tries to compile against both vtk6 and vtk7 and fails to link against vtk6.

2) I was able to compile director with vtk6.2 and qt5, but directorPython fails to draw the view port with an error mentioning AutoInit within a vtk "vtkRenderingFreeTypeFontConfig_AutoInit_Destructv"

I think my options are:
A) make ddApp compile against both vtk6 and vtk7-qt5. I don't know if that's possible.
B) Use vtk5-qt4 and the depreciated branch you created. I don't know if that would work with pcl and vtk6
C) Use vtk6 and fix the run time issue above. But you seem to suggest its a bad release.

Which approach would you suggest?

patmarion, 2018-01-12:
I would have expected the vtk6.2 failure to be a cmake or build failure.  If it actually builds and runs, but crashes with a runtime error, then it very well may be possible to just fix that run time error.

It could be something as simple as adding/removing some libraries from this list, like, what happens if you add "vtkRenderingFreeTypeFontConfig"?

https://github.com/RobotLocomotion/director/blob/master/src/vtk/DRCFilters/CMakeLists.txt#L60

mauricefallon, 2018-01-14:
Success! It took a day, but I found the issue.

In June @sankhesh added a commit to [ddQVTKWidgetView.cpp](https://github.com/RobotLocomotion/director/blob/master/src/app/ddQVTKWidgetView.cpp#L104)

Here:
https://github.com/RobotLocomotion/director/commit/dd4ee5d12ebbf505279d4248f2530ce5fc87c3b4#diff-ab3b6e434c7408392ed6672f3ef053d8

Basically when I forced the use of the older code director works just fine. Such that those lines read:
```
  layout->addWidget(this->Internal->VTKWidget);
  this->Internal->VTKWidget->SetUseTDx(true);
  this->Internal->RenderWindow = vtkSmartPointer<vtkRenderWindow>::New();
  this->Internal->VTKWidget->SetRenderWindow(this->Internal->RenderWindow);
  this->Internal->RenderWindow->SetMultiSamples(8);
```
The result is the full director drc build works within a ROS/catkin build system with vtk6 and qt5.

I'm not familiar with what's going on here, any idea what's going on?

sankhesh, 2018-01-15:
@mauricefallon VTK 6.2 does not support Qt 5. It might compile but there are known rendering issues. See http://vtk.1045678.n5.nabble.com/Depth-peeling-and-Qt-td5746110.html for example.

It might just be a matter of fixing director to fail gracefully if the supported versions are not available.