RobotLocomotion/director #579: vtk-related build error

grossmanlev, 2018-01-11:
Hello,
I am currently trying to install Director onto an NVIDIA Jetson TX2 running Ubuntu 16.04.  However, after running `make superbuild`, I am coming across this error:

```[  0%] Generating depth_compose.cxx
../../../../../install/bin/vtkEncodeString-7.1: 1: ../../../../../install/bin/vtkEncodeString-7.1: Syntax error: end of file unexpected
src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/build.make:73: recipe for target 'src/vtk/DRCFilters/depth_compose.cxx' failed
make[6]: *** [src/vtk/DRCFilters/depth_compose.cxx] Error 2
make[6]: Leaving directory '/home/nvidia/dev/director/build/src/director-build'
CMakeFiles/Makefile2:1123: recipe for target 'src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all' failed
make[5]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
make[5]: Leaving directory '/home/nvidia/dev/director/build/src/director-build'
Makefile:138: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/nvidia/dev/director/build/src/director-build'
CMakeFiles/director.dir/build.make:116: recipe for target 'src/director-stamp/director-build' failed
make[3]: *** [src/director-stamp/director-build] Error 2
make[3]: Leaving directory '/home/nvidia/dev/director/build'
CMakeFiles/Makefile2:71: recipe for target 'CMakeFiles/director.dir/all' failed
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[2]: Leaving directory '/home/nvidia/dev/director/build'
Makefile:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/nvidia/dev/director/build'
Makefile:22: recipe for target 'superbuild' failed
make: *** [superbuild] Error 2
```

Any help would be greatly appreciated!

patmarion, 2018-01-11:
Hmm, I haven't actually encountered this particular error before.

First, you could try using pre-compiled binaries that are known to work on Ubuntu 16:

https://bintray.com/patmarion/director/director/0.1.0-234-g74cea84#files

But if you need to compile for your architecture, one quick thing you can try to do is work around the issue by removing this particular build step from the CMakeLists.txt.  You could edit this file and simply remove everything from line 41 through line 55:

https://github.com/RobotLocomotion/director/blob/master/src/vtk/DRCFilters/CMakeLists.txt

grossmanlev, 2018-01-12:
Thanks for the response.  Unfortunately due to Jetson's system architecture, the binaries do not seem to run.  I have tried commenting out the lines from the CMakeLists.txt file, but this also returns an error when compiling:

```
[  1%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkDepthImageUtils.cxx.o
[  1%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkGridSource.cxx.o
[  2%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkFrameWidget.cxx.o
[  2%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkFrameWidgetRepresentation.cxx.o
[  2%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkInteractorStyleTerrain2.cxx.o
[  3%] Building CXX object src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkPickCenteredInteractorStyle.cxx.o
make[6]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libSM.so', needed by 'lib/libvtkDRCFilters.so'.  Stop.
make[6]: Leaving directory '/home/nvidia/dev/director/build/src/director-build'
CMakeFiles/Makefile2:1123: recipe for target 'src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all' failed
make[5]: *** [src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/all] Error 2
make[5]: Leaving directory '/home/nvidia/dev/director/build/src/director-build'
Makefile:138: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/nvidia/dev/director/build/src/director-build'
CMakeFiles/director.dir/build.make:116: recipe for target 'src/director-stamp/director-build' failed
make[3]: *** [src/director-stamp/director-build] Error 2
make[3]: Leaving directory '/home/nvidia/dev/director/build'
CMakeFiles/Makefile2:71: recipe for target 'CMakeFiles/director.dir/all' failed
make[2]: *** [CMakeFiles/director.dir/all] Error 2
make[2]: Leaving directory '/home/nvidia/dev/director/build'
Makefile:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/nvidia/dev/director/build'
Makefile:22: recipe for target 'superbuild' failed
make: *** [superbuild] Error 2

```